MarlinFirmware/Marlin #14435: [BUG] Freezes while printing (SKRv1.3 + MKS TFT28 + TMC2208 + UART)

LastDragon-ru, 2019-06-28:
### Description & Steps to Reproduce

After a short time after start printing Marlin 2.0 (93cd66ac1146603e8a9911fceb3c63f4351f4d8e) freezes approx for a minute, then continue printing, stop again, etc (please see https://youtu.be/J9v7T9czITQ). Board doesn't respond via USB while freeze. M122 returns "ok". Calibration cube prints perfectly with the same settings. 

### Additional Information

* Dependency Graph: `<Servo> 1.0.0`, `<LiquidCrystal> 1.0.0`, `<U8glib-HAL> 0.4`, `<TMCStepper> 0.4.5`, `<Adafruit NeoPixel> 1.1.8`, `<SailfishLCD> #e35d62c`
* Config: [Configuration.h](https://github.com/LastDragon-ru/Marlin/blob/TevoLittleMonster-SKRv1.3/Marlin/Configuration.h), [Configuration_adv.h](https://github.com/LastDragon-ru/Marlin/blob/TevoLittleMonster-SKRv1.3/Marlin/Configuration_adv.h)
* [CFFFP_Body2.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/3340142/CFFFP_Body2.zip)
* `M122`:
    ```
    G:\3D\Marlin>pio device monitor --echo
    --- Miniterm on COM11  250000,8,N,1 ---
    --- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
    M122
                    X       Y       Z       E
    Enabled         true    true    true    true
    Set current     1000    1000    1000    1000
    RMS current     994     994     994     994
    MAX current     1402    1402    1402    1402
    Run current     17/31   17/31   17/31   17/31
    Hold current    8/31    8/31    8/31    8/31
    CS actual               17/31   17/31   17/31   8/31
    PWM scale       44      49      45      24
    vsense          0=.325  0=.325  0=.325  0=.325
    stealthChop     true    true    true    true
    msteps          32      32      32      16
    tstep           max     max     max     max
    pwm
    threshold               79      79      79      15
    [mm/s]          125     125     125     126
    OT prewarn      false   false   false   false
    off time                4       4       4       4
    blank time      24      24      24      24
    hysteresis
    -end            2       2       2       2
    -start          1       1       1       1
    Stallguard thrs
    DRVSTATUS       X       Y       Z       E
    stst            *       *       *       *
    olb
    ola
    s2gb
    s2ga
    otpw
    ot
    157C
    150C
    143C
    120C
    s2vsa
    s2vsb
    Driver registers:
                    X       0xC0:11:00:00
                    Y       0xC0:11:00:00
                    Z       0xC0:11:00:00
                    E       0xC0:08:00:00

    Testing X connection... OK
    Testing Y connection... OK
    Testing Z connection... OK
    Testing E connection... OK
    ok

    --- exit ---
    ```

LongLiveCHIEF, 2019-06-28:
Can you share your print speed settings?  I'm wondering if they're freezing when hybrid mode kicks in.

LastDragon-ru, 2019-06-29:
I also tried

1) Disable StealthChop by `M569 S0 X Y Z` - same
2) Disable `SQUARE_WAVE_STEPPING`, `S_CURVE_ACCELERATION`, `ADAPTIVE_STEP_SMOOTHING` - same
3) Decrease `BLOCK_BUFFER_SIZE` from 32 to 16 - same
4) New microsd for MKS TFT28 - same
5) Disable `LIN_ADVANCE` - same

<details>
<summary>Settings (Cura 4.10):</summary>

[image]
[image]

</details>

teemuatlut, 2019-06-29:
Does turning off monitoring do anything?

LastDragon-ru, 2019-06-30:
@teemuatlut, do you mean `MONITOR_DRIVER_STATUS`? It was disabled. I enabled it, connect serial port to usb (in parallel with tft28 with is why there are 2 logs, sorry, I don't know how to combine them...) and start printing (with `M569 S0 X Y Z E`). 

In output I see two errors like

> Error:Line Number is not Last Line Number+1, Last Line: 1
> Resend: 2

don't know what does it mean... but probably they are not related to my problem.

When printer halts I see only:

> echo:busy: processing
> echo:busy: processing
> echo:busy: processing

Gcodes before halt:

> N253 G1 F3000 E77.71414*53
> N254 G1 F17987547480000 Z1.3*17 (-> https://github.com/Ultimaker/Cura/issues/5952)
> N255 G0 F3600 X12.506 Y9.685 Z1.3*45
> N256 G0 X-42.904 Y-1.677*21

Seems this is not TMC bug... :(

<details>
<summary>Marlin output</summary>

```
>pio device monitor --echo --baud=115200
--- Miniterm on COM10  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

echo:LPC1768 (100Mhz) Initialized
start
echo:PowerUp
Marlin bugfix-2.0.x

echo: Last Updated: 2019-06-26 | Author: (none, default config)
echo:Compiled: Jun 30 2019
echo: Free Memory: 17479  PlannerBufferBytes: 3456
echo:V67 stored settings retrieved (646 bytes; crc 42443)
Unified Bed Leveling System v1.01 active

Unified Bed Leveling initialized.

Mesh loaded from slot 1
Mesh 1 loaded from storage.
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X160.00 Y160.00 Z160.00 E418.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z300.00 E50.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X4000.00 Y4000.00 Z4000.00 E4000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P2000.00 R4000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z10.00 E5.00
echo:Unified Bed Leveling:
echo:  M420 S1 Z10.00

Unified Bed Leveling System v1.01 active

Active Mesh Slot: 1
EEPROM can hold 3 meshes.

[... 741 lines cut ...]
N245 G1 F2400 X-11.204 Y10.238 E81.58179*25
N246 G0 F3600 X-11.768 Y10.238*103
N247 G1 F2400 X-12.436 Y9.569 E81.62884*38
N248 G0 F3600 X-12.436 Y10.134*109
N249 G1 F2400 X-12.332 Y10.238 E81.63616*20
N250 G0 F3600 X11.328 Y8.507*127
M105
N251 G1 F2400 X12.436 Y9.615 E81.71414*3
N252 G1 X12.506 Y9.685*30
N253 G1 F3000 E77.71414*53
N254 G1 F17987547480000 Z1.3*17
N255 G0 F3600 X12.506 Y9.685 Z1.3*45
N256 G0 X-42.904 Y-1.677*21
M116
M104 S0
M140 S0
G91
G1 Z5
G90
G28 X0 Y0
M84
M107
N-1 M110*15
```
</details>

LastDragon-ru, 2019-06-30:
> Seems this is not TMC bug

As described in https://github.com/Ultimaker/Cura/issues/5753 I set "Maximum z speed" and it works! 🎉 

But this also means that Marlin incorrectly handles huge `F` value ...

LastDragon-ru, 2019-06-30:
#14454

closed 2019-06-30 by LastDragon-ru