MarlinFirmware/Marlin #14436: Long waiting time in bed leveling

sirtheta, 2019-06-28:
I just started a print with a build from today. It is my first one with Marlin 2.0
I got an issue in bed leveling (the issue also appears while homing Z). The printer waits very long for every probing and the display is reading "Bed heating" all the time. It seems that something like a hysteresis for the bed temperature is not set correctly. (I checked TEMP_BED_WINDOW          and TEMP_BED_HYSTERESIS, the values are set to standard) 
Here is what I do when I start a print:
```
M140 S60   ; set bed Temp
M190 S60   ; wait for bed Temp
G28        ; Home all axes
G29        ; Bed Leveling
```
I am using a BLTouch for probing. My printer is a Ender 3 with a MKS Gen L and TMC2130 steppers. I have multiple probing activated and I am using the AUTO_BED_LEVELING_BILINEAR function.

mylife4aiurr, 2019-06-28:
I having been reporting G29 bilinear is broken on Delta's for days. Seems maybe it's broken for all. On Delta's the error is:
Auto Bed Leveling: BILINEAR (disabled)
: ? (L,R,F,B) out of bounds.

What error do you get? Does it eventually work?

I have cross referenced other similar issues, I have made them aware on discord, I donated, I've become patreon, gave detailed reports with configs and installed several different builds.

sirtheta, 2019-06-28:
I get no error. The only problem is that I have to wait very long until the bed leveling is done.

thinkyhead, 2019-07-03:
> I having been reporting G29 bilinear is broken on Delta's for days

Sorry you haven't been put at the front of the line, but we are all volunteers working on Marlin in our spare time.

mylife4aiurr, 2019-07-03:
i understand every1 volunteers beleive me. it wasn't that i wanted to be 1st :)
it was that delta's dont seems to get same level of support and prechecks and that I'd been ringing the alarm without acknowlegement. i write this response b4 you said you'd look into it.

sirtheta, 2019-07-04:
I just did another test after pulling the last 2.0 branch today. The bed leveling works perfect when the bed heating is not on (seting the temperature with M190 or M140 makes no difference). What I can observe is that the BLTouch is not deploying as normal after probing and move back. It seems that the printer is waiting for something.

thinkyhead, 2019-07-05:
Let's see about getting more details….

- Download the latest `bugfix-2.0.x`
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. Go only as far as you can and don't let it hurt itself!
- Copy the log output into a **.TXT file** and attach it to your next reply.

From the log we should hopefully get a better idea of what's going on.

thinkyhead, 2019-07-05:
> i wrote this response b4 you said you'd look into it.

Thanks, @mylife4aiurr — We'll figure it out at some point I am sure! Remember that you should not be setting any square boundaries in your own config. Those are automated, but if there's even a small error in the equation then the calculated square bounds could be off a tiny bit. AFAIK we do add some margin to account for the potential error, as we have been over this code a few times. If you don't use LRFB in your own printing, then you can simply modify `G29` in the mean-time to get rid of that check or to add a wider boundary. That workaround will be guaranteed to work while we're all busy with other tasks.

mylife4aiurr, 2019-07-06:
G33 working again.
https://github.com/MarlinFirmware/Marlin/issues/14347

```
#define BLTOUCH
#define X_PROBE_OFFSET_FROM_EXTRUDER  9   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -35   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER  0   // Z offset: -below +above  [the nozzle]
#define AUTO_BED_LEVELING_BILINEAR

#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
// Set the boundaries for probing (where the probe can reach).
#define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
#define RIGHT_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
#define FRONT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE))
#define BACK_PROBE_BED_POSITION DELTA_PRINTABLE_RADIUS -(MIN_PROBE_EDGE)
```
So as defined above you'll be able to compile fine, then at time of 1st g29 error :(
G29 ? (L,R,F,B) out of bounds. 

I was able to get it working by gradually increasing each parameter
G29 L-10 R10 F10 B-10 success
G29 L-10 R60 F60 B-60 success
G29 L-80 R90 F60 B80 out of bounds

  settled on:
  #define LEFT_PROBE_BED_POSITION -80
  #define RIGHT_PROBE_BED_POSITION 83
  #define FRONT_PROBE_BED_POSITION -60
  #define BACK_PROBE_BED_POSITION 83

also had to invert my probe offsets
#define X_PROBE_OFFSET_FROM_EXTRUDER  -9   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 35   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER  0   // Z offset: -below +above  [the nozzle]
because delta is offset compared to Cartesian??, idk doing so worked

 *     +-- BACK ---+
 *     |           |
 *   L |    (+) P  | R <-- probe (20,20)
 *   E |           | I
 *   F | (-) N (+) | G <-- nozzle (10,10)
 *   T |           | H
 *     |    (-)    | T
 *     |           |
 *     O-- FRONT --+
 *   (0,0)

that diagram is so confusing for delta, wish there was a delta depiction

anyway im good to go @thinkyhead

sirtheta, 2019-07-06:
So I did a Bed Leveling with DEBUG activated. See atached files. First I did a bed leveling wit bed temp set to 60 as I do normally. The second log is with bed temp OFF. As said before, when bed temp is not set, everything works as it should.
[log_BedLevelingTempOFF.txt](https://github.com/MarlinFirmware/Marlin/files/3364453/log_BedLevelingTempOFF.txt)
 
[log_BedLevelingWithS60.txt](https://github.com/MarlinFirmware/Marlin/files/3364450/log_BedLevelingWithS60.txt)

sirtheta, 2019-07-06:
I just saw my mistake. It is so embarassing... I had  WAIT_FOR_BED_HEATER activated... I am sorry for that mistake. I close this issue.

closed 2019-07-06 by sirtheta