MarlinFirmware/Marlin #14445: [BUG] Error compiling ubl_G29.cpp when Z_PROBE_END_SCRIPT is defined

igorepi, 2019-06-29:
### Description

Latest Marlin bugfix-2.x does not compile when UBL is enabled and Z_PROBE_END_SCRIPT is defined. Compilation fails with message "'process_subcommands_now_P' was not declared in this scope" in Marlin\src\feature\bedlevel\ubl\ubl_G29.cpp line 682.

### Steps to Reproduce

Enable some probe, UBL and Z_PROBE_END_SCRIPT in Configuration.h:

#define FIX_MOUNTED_PROBE
#define AUTO_BED_LEVELING_UBL
#define RESTORE_LEVELING_AFTER_G28 false
#define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X15 Y330\nG1 Z0.5\nG1 Z10"

and try to build Marlin

**Expected behavior:** [What you expect to happen]

Marlin compiles successfully

**Actual behavior:** [What actually happens]

Marlin does not compile

#### Additional Information

I think that Marlin\src\feature\bedlevel\ubl\ubl_G29.cpp line 682 should be
  `gcode.process_subcommands_now_P(PSTR(Z_PROBE_END_SCRIPT));`
instead of 
  `process_subcommands_now_P(PSTR(Z_PROBE_END_SCRIPT));`

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3341661/Configuration.zip)

Ludy87, 2019-06-30:
Fix is online, you can close this issue

leytanelectronics2, 2019-06-30:
what was the issue for this? was it causing the gantry to move up X distance from the actual part or something?

closed 2019-07-02 by igorepi

igorepi, 2019-07-02:
@leytanelectronics2, it was just a typo in b961251899ff045a4d3da4d8e3816ed9876befd0

AnHardt, 2019-07-02:
It's not *just* a *typo*. That can always happen. It's a *systematic error* in programming / integration. It's not, at least test-compiling the changed code-path, at least, once. It's missing to verify a change. (Don't trust in Travis! Travis can't do that job for the programmer. Testing all possible combinations of features would take forever. But the programmer usually exactly knows how to switch on the just changed code-path - else he shouldn't have changed the code in there. )

oneofnine, 2019-07-02:
> 
> 
> But the programmer usually exactly knows how to switch on the just changed code-path - else **_he_** shouldn't have changed the code in there.

-he +they ;)

thinkyhead, 2019-07-03:
We all admire and want perfection. But there is a very good reason we have a disclaimer. We experiment, we rush, we get tired, and mistakes fall through. Hence we use a VCS that allows reverting.

Some may feel that every typo or error is a _**disaster**_. But with 1400 unread notifications to get through, I simply don't have time to beat myself about the face and neck. While I am working on improving tools and processes, my main focus right now is integrating as many needed patches as possible, and mistakes will be made.

AnHardt, 2019-07-03:
Obviously **not every** typo is a disaster. It's not in comments, not in new option names, variables or function - when consistently "wrong". 
Introducing single typos detectable by compiling the right code-path is not a disaster but at least annoying when found by a user after merging.  This will cause in average two new issues (because the average user does not follow any other issues) and (if all goes well only) one PR to fix. So in average 3 more threads to read for you - growing your unread notifications - lessening your time you are able to invest for test-compiling something - causing more bugs to slip threw.
Typos changing the logic, like typing a '<' instead of a '>' in a compare are usually a disaster when the code is not tested if it does what is intended to do - before a merge. That kind of typos can survive for years if only used in rarely used features or combinations if not purposely tested by the author. If found too late probably the author left the team, and nobody als can remember what the code should have done.

**I** do not request **you** (@thinkyhead ) to test functionality of every PR. But I beg you for a  test-compile of all obviously affected code-paths of code you changed in PRs - and functionality testing your own code. Alternatively you could make PRs to the sources of the PRs you changed, instead of pushing them - giving back the duty for testing to the original PRs authors.

How about some temporarily added Travis test, removed before merging?

Due to the high amount of (unnecessary) errors Marlins reputation is that low, that even I begin to ask: "Where in Marlin is the error?", before i begin to check my hardware (what is not that unlikely to fail with kids and cats in the house) or configs. Causing more (unnecessary) issues.

How about sharing your workload by increasing the team? Oh, i see what the problem with this is. (Not able, not willing, no time, different focus, different philosophy,  ... and increased workload for coordination (in no special order))

Ludy87, 2019-07-03:
Marlin 2 is in the Alpha Stadium, so mistakes have to happen, which should not happen when publishing.

AnHardt, 2019-07-04:
We claim Malin bugfix-2.0.x to be the best Marlin we have - better than all the "stable" releases. Just the 32 bit processor part is "beta" for the most ones now.

thinkyhead, 2019-07-04:
Both are true. It is better, but it is also in flux. We don't have a release or "beta-but-not-exploding" branch as a separate download for 2.0.x at this time, but once we have a release it will be more sensible for people to trust the download link. Right now if you download 2.0.x you have no idea whether it will even compile, because as mentioned, Travis doesn't test every feature exhaustively.

I would love it if Travis did test every feature exhaustively. Then we would catch more of the obvious oopsies. Perhaps someone who cares a lot about bugs getting missed will step up and really improve the TESTS files in `buildroot`! I would do so right this minute, but my list is long….

Roxy-3D, 2019-07-04:
The problem comes down to balancing the goals against the available resources.     We literally have thousands of people using Marlin firmware.   Some of the people using Marlin want it to be 'State of the Art' and 'Pushing the Technological Envelope'.    Others want it to be 100% bug free.   Other people want the code perfectly clean and to be a stellar example of Object Orientated Programming.   

At a higher level...   The reason people contribute their time, effort and expertise to Marlin is because they enjoy doing that.    Nobody is getting paid for helping move Marlin forward.   It is 100% volunteer effort.

I have no way to test this hypothesis.   But I don't think it would matter how perfectly bug free the code was.   It wouldn't matter how well it supported new technology.   Even if it was 100% bug free...     There would be people complaining that we were not doing releases fast enough.   Or that the release notes were incomplete and should include easier to follow steps on this or that.

PR #14447 (2019-06-30): Fix compile error with Z_PROBE_END_SCRIPT
Fixes #14445