MarlinFirmware/Marlin #14448: [FR] Add Drilling canned cycles to Marlin 2.0

FNeo31, 2019-06-29:
Like I talk with @thinkyhead  on Discord,
I'm sharing the implementation of Drilling Canned Cycles (G81, G82 and G83)

On Last Bugfix from today just added the call of the cycles on gcode.cpp and gcode.h

add variable on adv cfg:

/**
 * Drilling canned cycles
 * 
 * add G81, G82 and G83
 * 
 */
#define DRILLING_CANNED_CYCLES

and add the attached files of functions.

This drilling cycles aren't optimized and on patterns you need to put the position of new hole and ask for cycle again. This isn't corrected because I don't have too much time to check a solution (one of the reason is the G80 be in use now for other functions).

Wasn't perfect but it's a beginning. :) 

[drilling cycles.zip](https://github.com/MarlinFirmware/Marlin/files/3342237/drilling.cycles.zip)

EvilGremlin, 2019-07-13:
Maybe post it by pull request?

boelle, 2019-07-21:
Thanks for your contribution! Please send as a PR
to the `bugfix-2.0.x` branch and be careful to target
`bugfix-2.0.x` when resubmitting the PR.