MarlinFirmware/Marlin #14449: power loss recovery Z axis jumps up 1CM from print after re-homing

leytanelectronics2, 2019-06-29:
I have had the power loss recovery option activated in my copy of marlin but when it re_homes and goes to continue the print head jumps up 1CM from the actual print before continuing, anyone have any theories as to why this is happening and any options on how to fix this?

boelle, 2019-07-21:
Please test with the [`bugfix-2.0.x` branch](https://github.com/
MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) so we're all on the same
page here. That branch contains all the latest patches.

boelle, 2019-09-24:
@leytanelectronics2 is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle