MarlinFirmware/Marlin #14450: [BUG] (short description)

RudyBenner, 2019-06-29:
For a few hours, since I suspect the platformio.ini update, strange things have been happening. .pioenv fills up with folders for every one of the selections. I don't even know how to describe this.

I have deleted all the usual stuff. It comes right back. Rebased. Am I the only one?

rudy

thisiskeithb, 2019-06-29:
Does it compile successfully if you change `default_envs` to `env_default` in `platformio.ini`?

There was a change to that file 16 hours ago: https://github.com/MarlinFirmware/Marlin/commit/6550a222aab41d31c7338cd4f5bfc27427110fab.

RudyBenner, 2019-06-29:
tried that, no joy.

RudyBenner, 2019-06-29:
Is it working for you?

RudyBenner, 2019-06-30:
right now, it is unusable.

RudyBenner, 2019-06-30:
Fixed. Your suggestion was the logical choice, what I missed was the extra spaces on the previous lines.

[image]

RudyBenner, 2019-06-30:
Bug needs fixing yet.

RudyBenner, 2019-06-30:
I will leave my patch until it gets fixed.

closed 2019-06-30 by RudyBenner