MarlinFirmware/Marlin #14451: [Question] M100..everything ok?

GMagician, 2019-06-30:
I'm haveing some issues on the board I'm developping. First call works ok and seconds hang system.

From cheated M100 to dump memory and it seems someone is using heap (that's my problem to discover who is), but looking inside code I found:

```c++
  for (uint8_t i = 0; i < 16; i++) {
        char ccc = (char)start_free_memory[i]; // cast to char before automatically casting to char on assignment, in case the compiler is broken
        if (&start_free_memory[i] >= (char*)queue.buffer && &start_free_memory[i] < (char*)queue.buffer + sizeof(queue.buffer)) { // Print out ASCII in the command buffer area
          if (!WITHIN(ccc, ' ', 0x7E)) ccc = ' ';
        }
```

but AFAIK queue.command resides in BSS area that's outside free memory area....so why this is present?

@Roxy-3D may help me since she wrote such code, I think

GMagician, 2019-06-30:
I think that heap is used by usb stack...maybe some change on M100 to change start free memory depending on used heap?

Roxy-3D, 2019-06-30:
M100 was written for the 8-bit AVR's.   They have a memory map that looks like:

[image]

Pretty much...  If you are not doing malloc()'s and free()'s, the amount of free memory is between the top of the stack and the end of the global variables.      Most GCC implementations follow a similar scheme.    If you can find a picture of what your 32-bit processor's memory map looks like, I'll help you get M100 verified as correct on your platform.

GMagician, 2019-06-30:
I know that my MPU uses the same approach and I'm almost sure usb stack uses malloc that's with M100 hang it. Maybe some heap walk to find real value for 'start_free_memory'.
From your map you confirm my suspect. queue.buffer is static variable and is stored in bss, so you will never find it in code above.

GMagician, 2019-06-30:
avr should use __brkval to identify free memory, otherwise a hang may occurs (is avr code doesn't use  heap but other cpu?)

Roxy-3D, 2019-06-30:
The 8-bit AVR code used to use __brkval.    I don't know the specifics of why it was changed.   But I think the change was an attempt to make the code more portable for the 32-bit processors.

With regard to the USB code...   If it is using malloc() we probably want to change that.   We spent some time and effort removing all malloc()'s (and free()'s) from the code for reliability reasons.

If there are no malloc()'s or free()'s, it becomes much easier to get an accurate free memory value.

GMagician, 2019-07-01:
> The 8-bit AVR code used to use __brkval. I don't know the specifics of why it was changed. But I think the change was an attempt to make the code more portable for the 32-bit processors.

That's no more this way for avr and I don't understand why it have been changed (maybe because all malloc have been removed so __brkval matches heaplimit), not for sure for 32 bits since there is a #if specific for avr to get start and end of available ram. At the moment I solved in my Grand Central (SAMD51) with _sbrk(0) to get end of heap. Now that I use it (with a 0x200 cushion barrier), M100 works perfectly.

> With regard to the USB code... If it is using malloc() we probably want to change that. We spent some time and effort removing all malloc()'s (and free()'s) from the code for reliability reasons.

That's a little bit more complicated. SAMD51 implements a USB stack in its "arduino" framework. D+ and D- signals are handled directly by code and if you want to remove "malloc" you have to rewrite whole usb stack...That's too much at the moment. USB com is working, M100 is working and I need to implement eeprom, SD, LCD, fastpwm, emergency parser before. So a lot of job before I am able to think to mess up already working code.

closed 2019-07-01 by GMagician

Roxy-3D, 2019-07-01:
> At the moment I solved in my Grand Central (SAMD51) with _sbrk(0) to get end of heap. Now that I use it (with a 0x200 cushion barrier), M100 works perfectly.

Can you provide the changes you did to get your processor running?

GMagician, 2019-07-01:
M100.cpp:
```c++
#elif defined(__SAMD51__)

  extern unsigned int __bss_end__, __StackLimit, __HeapLimit;
  extern "C" void * _sbrk(int incr);

  void *end_bss = &__bss_end__,
       *stacklimit = &__StackLimit,
       *heaplimit = &__HeapLimit;

  #define MEMORY_END_CORRECTION 0x400

  char *free_memory_start = (char *)_sbrk(0) + 0x200,     //  Leave some heap space
  *free_memory_end = (char *)stacklimit - MEMORY_END_CORRECTION;
```

That's not absolutely safe. I checked usb code and it seems there is only one "new" and "delete" in code but it can take up to 256*8 bytes (is used for usb end point allocation)

Roxy-3D, 2019-07-02:
> and it seems there is only one "new" and "delete" in code but it can take up to 256*8 bytes (is used for usb end point allocation)

OK.   I understand.

AnHardt, 2019-07-02:
When the library code for "new" and "delete" is in the PROM anyway one could just malloc() a bit of heap to get its current end address.
If we are not interested in the amount of remaining stack but just in, if heap and stack are about to collide, one could use this new (last) heap area for the test if anything there in changed.

GMagician, 2019-07-02:
@AnHardt to get end of currently used heap I used _sbrk(0). malloc + free is a univesal/standard way to do that.
The results are identical, don't know if this requires changes.
Please note that _sbrk(0) is also used in other HAL to return free memory on startup. If changed here maybe should be uniformed in all HALs

AnHardt, 2019-07-02:
Hmm.
On the systems not using the heap it's wasteful to pull in the heap management code just the find the start of the zero length heap. If we have that address from a linker constant i'd prefer to use that.

GMagician, 2019-07-02:
One question. not directly related to this, in adafruit usb stack new and delete are used. Forcing 8 fixed 256 bytes array may remove heap use but...To be "officially" supported adafruit has to change it, platformio need to use such modified code and finally we use it. Other solution will be to use a different "patched" framework for Marlin...What is "normal" action Marlin programmer should take to solve this?

Roxy-3D, 2019-07-14:
If you are on a 32-bit platform...  Burn the memory.   Marlin still works with an 8-bit AVR having 8KB of RAM memory on the AVR's.

But it is very possible we have some clean up work to do on the 32-bit platforms.   With that said... Burning some memory isn't going to hurt you on those board.

GMagician, 2019-07-14:
Adafruit is 32 bits platform. And in a deep look is not 256 bytes. It seems 8 class instances so harder to remove