MarlinFirmware/Marlin #14452: [FR] Better granularity for Z moves

yet-another-average-joe, 2019-06-30:
Hello, and thanks for this great software !

I find something really frustrating. When adjusting the Z-Offset, the granularity is 0.1mm. 0.1mm is half the thickness of a 1st layer for PLA and a 0.4mm nozzle.

0.05 or 0.025mm would be better.
0.05mm is one step with a 200steps/rev stepper, 0.025 is 1/2 step, with 8mm leaedscrews.

I just tested 0.025, and it makes things really easier when using feeler gauges. Much easier than moving using a terminal and G1 commands, or measuring a 1st layer).

It is easy to add (I did it...)

In menu_motion.cpp :

`void menu_move_0025mm() { _goto_manual_move( 0.025f); }`

and :

`MENU_ITEM(submenu, MSG_MOVE_0025MM, menu_move_0025mm);`

And in langage_en.h :

```
#ifndef MSG_MOVE_0025MM
  #define MSG_MOVE_0025MM                       _UxGT("Move 0.025mm")
#endif
```

Maybe, some will say that such a granularity makes no sense. This is untrue. I measured repeatability on my printer using a 0.01mm Mitutoyo DTI, and then a 0.002mm (2 microns) one, (Compac, swiss made). Repeatability is within +-0.002 mm (+-2 microns !) (I know, repeatability is NOT precision).

Is there any chance that such an enhancement could be implemented in a future release ? It really makes z-offset adjustments much easier. Until these changes in code, Z offset was a pain ! (and implied measuring a 1st layer using a micrometer)

Adding macros in langage_XX.h files is no big deal, and even with no knowledge of other langages, it could be done within a few minutes.

Ludy87, 2019-06-30:
enable `BABYSTEPPING` and `BABYSTEP_ZPROBE_OFFSET`

yet-another-average-joe, 2019-06-30:
Babystepping cannot be used except while printing, 
M851 or Z offset in the configuration submenu are the same (followed with M500 or save settings). Z axis has to be moved in order to adjust the position, and 0.1mm steps are too much. 
5 lines of code solve the problem, adding a line in the menu.

[image]

thisiskeithb, 2019-07-01:
> Babystepping cannot be used except while printing,

Enabling `BABYSTEP_ALWAYS_AVAILABLE` should.

yet-another-average-joe, 2019-07-01:
@Ludy87  and @thisiskeithb, thank you, I was missing this !

It was activated. I also had DOUBLECLICK_FOR_Z_BABYSTEPPING and BABYSTEP_ALWAYS_AVAILABLE activated, but "cold" babystepping pops up when doing a long click not a double click. Weird. (don't know for "hot" babystepping, as i've not been using it for a while)

This is the reason why I added the menu item and was about to issue a pull request. This being sayed, I think that "move axis 0.025" is not 100% useless. I will keep it in my sources. It doesn't hurt.

For some reason, I was also missing BABYSTEP_ZPROBE_OFFSET (never payed attention, why ???)

tpruvot, 2019-07-01:
i had that made on my fork, made PR #14456 which do that

yet-another-average-joe, 2019-07-01:
Great !

closed 2019-07-06 by yet-another-average-joe