MarlinFirmware/Marlin #14453: Z safe homing is not working

joaomamede, 2019-06-30:
Yesterday I updated to the latest version of Marlin 2.0.x.
(this was working before)

Now my Z_SAFE_HOMING is not working

When I do a G28, it correctly does the X, then Y, but now the Z is not done in the center of bed (I have bltouch) and the speed/noise from the Z axis is much higher.
My nozzle hit the bed pretty strongly on my first try (the bltouch still triggers the Zmin if I do it with my finger)

Here are my configuration files.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/3343184/Configurations.zip)

Was there anything changed recently on the Z_SAFE_HOMING?

Ludy87, 2019-06-30:
```c++
// The size of the print bed
#define X_BED_SIZE 235
#define Y_BED_SIZE 235

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS -5
#define X_MAX_POS X_BED_SIZE+18
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 250
```

change to:

```c++
// The size of the print bed
#define X_BED_SIZE 220
#define Y_BED_SIZE 220

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS -5
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 250
```

joaomamede, 2019-06-30:
That's not the issue. I found it. It's the interpolation and X Y Z E microsteps that are moving the motors in a wrong way. I'm closing (remove if possible)

closed 2019-06-30 by joaomamede

Ludy87, 2019-06-30:
then you have to describe the problem correctly