MarlinFirmware/Marlin #14454: [BUG] Incorrect processing F17987547480000 (halt/freeze)

LastDragon-ru, 2019-06-30:
### Description

By default Cura 4.1 generates invalid gcode with huge `F17987547480000` values (see https://github.com/Ultimaker/Cura/issues/5753), this halts printer (actually it moving but very very very very slowly).

### Steps to Reproduce

> G1 F17987547480000 Z1.317

**Expected behavior:** 

Probably `F` should be set to "max value" from config.

#### Additional Information

* #14435 (initial issue + gcode)

Ludy87, 2019-06-30:
@thinkyhead, would change be the right way to stop these behaviors?

`gcode.cpp`

```c++
  if (parser.linearval('F') > 0) {
    const float max_default_feedrate[] = DEFAULT_MAX_FEEDRATE;
    LOOP_XYZE(i) {
      if (parser.value_feedrate() <= max_default_feedrate[i])
        feedrate_mm_s = MMM_TO_MMS(parser.value_feedrate());
      else
        feedrate_mm_s = max_default_feedrate[i];
    }
  }
```

thisiskeithb, 2019-06-30:
Related PR: https://github.com/MarlinFirmware/Marlin/pull/13929

oneofnine, 2019-07-12:
The trouble is, Marlin crashes and reboots without warning when unintended (or intended) extreme GCode parameters are fed to it .. this is not good for any firmware or any software, you must do sanity checking of some sorts to prevent it doing so, it should at least protect itself and the machine.

Patag, 2019-07-12:
https://github.com/Ultimaker/Cura/issues/5753

LongLiveCHIEF, 2019-07-14:
For what it is worth, the `Maximum Z Speed` tooltip in Cura for this setting reads:

```
The maximum speed with which the build plate is moved. Setting this to zero causes the print to use 
the firmware defaults for the maximum z speed.
```

boelle, 2019-09-24:
@LastDragon-ru is the issue still there?

LastDragon-ru, 2019-09-24:
@boelle, #13929 not yet merged -> issue still actual.

boelle, 2019-10-13:
@LastDragon-ru it was merged 11 days ago, you think it has fixed it?

LastDragon-ru, 2019-10-13:
Actually, not tested yet... But I think we can close it :)

closed 2019-10-13 by LastDragon-ru

boelle, 2019-10-13:
oki, let us know if we need to reopen

InsanityAutomation, 2019-10-15:
Its unlikely to be completely fixed. Im not sure if the overflow was in the planner handling (would be fixed) or the parser (not changed). So id call it 50/50 right now. Cura has removed this so instances should be less likely.