MarlinFirmware/Marlin #14459: [FR] Menu options to override fan speed and acceleration/jerk gcodes

MarkMan0, 2019-07-01:
Cura uses M205 to set different jerk values during the print. If I remember correctly, earlier versions used M201 to set acceleration. Now imagine that you've set them too high in the gcode. You can't change them in the menu, because they will get overwritten.
Pruse Slicer has a nice auto-fan feature. Imagine you've started a print, it's been going great, but now a steep overhang is coming. There is no way to turn on the fan, because the gcode will override the settings anyway, and turn off the fan. So a better option would be, to set a minimum fan speed in the menu.
Or you realised, that you can't use the fan, for example you're printing ABS, and you left the fan on.

I think it could be implemented like the tune menu, only visible during print, and when starting a print, it gets reset. Inside the *gcode*.cpp, it would check against these overrides. The values could be stored in a class, in a global fashion. I haven't looked at the 2.0 code yet, to determine the best way. 

Inside this new menu, there would be options, to limit the maximum/minimum acceptable values for acceleration, jerk and fan speed. Maybe more, like temperature or max feedrate.

This could save many hours and half complete prints.

InsanityAutomation, 2019-10-15:
I would consider this complete with the new clamping options that limit the max at compile time. It can no longer go over that badly. @boelle @shitcreek

boelle, 2019-10-15:
oki closing then, OP can always contact us

closed 2019-10-15 by boelle