MarlinFirmware/Marlin #14463: [BUG] [2.0.x bugfix] Crazy X moves

yet-another-average-joe, 2019-07-01:
### Description

X axis (and X axis only) moves 16x too much. Looks like the driver receives or interprets steps insteed of microsteps : 10mm -> 160mm, with crazy speed and acceleration

### Steps to Reproduce

1. Compile the last 2.0.x release
2. edit config files, or overwrite them with the last known good ones (I did it with the config files I successfully used with 2019-06-20 release
3. compile and flash
4. home X or move X axis

**Expected behavior:** [What you expect to happen]

asking for 10mm should move the head by 10mm

**Actual behavior:** [What actually happens]

160mm insteed of 10mm (0.1 insteed of 1, and 1 insteed of 10)

#### Additional Information

Printer : Tevo Tornado
Mobo : MKS Base v1.6
Drivers : A4982SLTP

I have been using Marlin 2.0.x for 1 year, without a problem.
Configuration files are upgraded using AkaJes marlin-configurator, and then verified/edited using WinMerge, NotePad++, Visual Studio or VisualStudioCode. Sources are compiled with marlin-Configurator (platformIO core) or VisualStusioCode (platformIO) or Visual Studio + Visual Micro. I flash with OctoPrint and the Firmware Updater plugin.

Thre was no problem with the release I downloaded on 2019-06-20 ; conf*.h files :

[2019-06-20.zip](https://github.com/MarlinFirmware/Marlin/files/3346510/2019-06-20.zip)

Today, I downloaded the new realease, upgraded/edited the conf files, compiled, flashed as usual : the X axis bugged. I verified everything, searching for some microstep value set to 1 insteed of 16, but no problem found. Then, I replaced the config files with the previous ones (2019-06-20), untouched. Same problem. Safe settings did'nt help. Cycling power has no effect.

[2019-07-01.zip](https://github.com/MarlinFirmware/Marlin/files/3346514/2019-07-01.zip)

Reflashing the 2019-06-20 version solves the problem.

This is why I think that the problem does not depend on the configuration files, and is not hardware related... I verified everything about 10 times.

yet-another-average-joe, 2019-07-02:
I probably found the solution to this problem.

When the Tornado was released with the MKS Base 1.5 /1.6 (A4982 drivers on my 1.6), the motherboard was set to BOARD_MKS_BASE in the Tevo firmware.

I never changed this parameter, and it ALWAYS worked fine from Tornado 1.0 (=Marlin 1.4) to Marlin 1.1.9 and Marlin 2.0.x until 2019-06-20... (I tested all versions, switching from 1.1.x to 2.0 from time to time)

After many hours trying to figure what happened, I discovered that BOARD_MKS_BASE_15 was introduced with Marlin 1.1.9 and 2.0 !

I just changed the motherboard declaration from BOARD_MKS_BASE to BOARD_MKS_BASE_15, and now the psychopathic behaviour of the X axis is gone ! Took me 7 hours browsing files, comparing them and compiling/flashing, but it seems it was worth it !

I suspect that other Tevo Tornado users will have the same problem when they will upgrade Marlin.

Could this be some recent bug in the legacy MKS_BASE interface ? Why the X axis ?

thinkyhead, 2019-07-11:
If you have a Tevo Tornado config you'd like to throw our way, we'll add it to the pile!

yet-another-average-joe, 2019-07-11:
My config files are in the first post, but my printer is heavily customized.

These configuration files will NOT fit a standard Tornado unless it is modded the same way.

**My configuration files could damage the Tevo hotend.**

thisiskeithb, 2019-07-11:
@yet-another-average-joe See PR https://github.com/MarlinFirmware/Marlin/pull/14576.

This config works on stock V1 & V2 hardware.

closed 2019-07-11 by yet-another-average-joe

yet-another-average-joe, 2019-07-11:
Do you mean that you flashed a Tornado using my configuration files, or do you mean that you released v 2.0.x firmware for standard Tornado's ?

Be aware that my Tornado is heavily customized (and therefore the conf*.h files) : quick swappable mirrors (with custom clips), the knockoff "buildtak" was removed (different PID parameters), the Tevo hotend has been replaced with a quick swap hotend system I designed for E3D hotends (V6 and Volcano), the X and Y endstops have been moved, Z endstop was removed, it uses a BLTouch, hotends PID parameters and Z-offset have to be set in the slicer start GCode depending on the hotend being used (I use eight different hotends), runout + tangle sensor, etc.

Do NOT flash a firmware with the configuration files I use ! They have to come with the related harware customizations, at this time still not on Thingiverse (coming soon with models and a video : a couple of months, I'm out of time at the moment).

thisiskeithb, 2019-07-11:
The configs in the PR have nothing to do with yours :)

yet-another-average-joe, 2019-07-11:
I feel better !

Your repo is a great idea.