MarlinFirmware/Marlin #14464: [BUG] TMC2209 Sensorless Homing not working

TheNitek, 2019-07-01:
### Description

I installed TMC2209 from Fysetc on my SKR 1.3 of my CoreXY and tried to get sensorless homing to work. UART connection works fine, but X and Y endstops always show as "TRIGGERED".
Jumpers on the SKR are closed any I ensured that there is an electrical connection between the DIAG pin and the endstop.

### Steps to Reproduce
Most recent Marlin 2.x bugfix
G28 XY

**Expected behavior:** Axis home

**Actual behavior:** Axis don't move, X/Y endstops are triggered

#### Additional Information
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3346644/Marlin.zip)

Ludy87, 2019-07-01:
try reversing * _ENDSTOP_INVERTING

TheNitek, 2019-07-01:
I tried, but the sanity check won't let me:
"SENSORLESS_HOMING requires X_MIN_ENDSTOP_INVERTING and ENDSTOPPULLUP_XMIN when homing to X_MIN."

Ludy87, 2019-07-01:
maybe @teemuatlut can say something, why the behavior does not correspond to the TMC2130

TheNitek, 2019-07-01:
In SanityCheck.h it says
```
  // Stall detection DIAG = HIGH : TMC2209
  // Stall detection DIAG = LOW  : TMC2130/TMC2160/TMC2660/TMC5130/TMC5160
  #define X_ENDSTOP_INVERTING !AXIS_DRIVER_TYPE(X,TMC2209)
```
so I guess this is as intended. TMC2209 says HIGH is triggered, so I guess the above should be correct

BioGeekJoey, 2019-07-01:
Does it work with sensorless homing uncommented in the config? Someone else on a BigTreeTech FB group was having issues with a CoreXY too.

TheNitek, 2019-07-02:
Just tried again and now the printer moves (not having a good sensitivity yet). I wonder if a power cycle was necessary. I'll try to investigate further ...

TheNitek, 2019-07-02:
Interesting: Resetting the SKR (via the reset button) does NOT reset the TRIGGER state of the TMCs. Right now my Y-Stepper is triggered forever

TheNitek, 2019-07-02:
Maybe this is similar to #8890 ?

BioGeekJoey, 2019-07-02:
I noticed that after a power cycle (actual unplug, not the reset button) all (any all everything, not just TMC driver settings) are set to 0 even though the correct values are stored in eeprom and loading the eeprom manually (M501 or via lcd) will fix it.

TheNitek, 2019-07-03:
I don‘t think this issue is related. After power on my settings are just fine.this is only about the TMCs

thinkyhead, 2019-07-03:
It's possible that the stepper init for 2209 is still missing a step. We'll have to see whether it's a problem with `TMCStepper` or within `tmc_util.*` in Marlin.

VOIDREP, 2019-07-03:
Hey, im having exactly the same issue except my endstops are now reading "open" on X/Y but the motor just stutters as if its sensitivity to sensorless homing is very low?

I have adjusted sensitivity from 0-255 and even taken the belt off my XY but the motor still just turns maybe 2/5mm before stopping on a home command?

this is very bizzare as i can read positive voltage from TMC2209 DIAG1 to Xmin signal pin?

running SKR 1.3 /TMC2209/ COREXY just as OP

heres my M122
14:29:08.452 : X Y E
14:29:08.452 : Address  1 1 1
14:29:08.452 : Enabled  false false false
14:29:08.452 : Set current 800 800 900
14:29:08.561 : RMS current 795 795 887
14:29:08.561 : MAX current 1121 1121 1251
14:29:08.561 : Run current 25/31 25/31 28/31
14:29:08.561 : Hold current 12/31 12/31 14/31
14:29:08.561 : CS actual  12/31 12/31 14/31
14:29:08.671 : PWM scale 35 31 16
14:29:08.671 : vsense  1=.18 1=.18 1=.18
14:29:08.671 : stealthChop true true false
14:29:08.671 : msteps  16 16 16
14:29:08.782 : tstep  max max max
14:29:08.782 : pwm
14:29:08.782 : threshold  29 29 271
14:29:08.782 : [mm/s]  123 123 30
14:29:08.782 : OT prewarn false false false
14:29:08.782 : OT prewarn has
14:29:08.782 : been triggered false false false
14:29:08.782 : off time  4 4 4
14:29:08.891 : blank time 24 24 24
14:29:08.891 : hysteresis
14:29:08.892 : -end  2 2 2
14:29:08.892 : -start  1 1 1
14:29:08.892 : Stallguard thrs 3 4 0
14:29:08.892 : DRVSTATUS X Y E
14:29:08.892 : stst  * * *
14:29:09.000 : olb
14:29:09.000 : ola
14:29:09.000 : s2gb
14:29:09.111 : s2ga
14:29:09.111 : otpw
14:29:09.111 : ot
14:29:09.111 : 157C
14:29:09.222 : 150C
14:29:09.222 : 143C
14:29:09.222 : 120C
14:29:09.222 : s2vsa
14:29:09.222 : s2vsb
14:29:09.222 : Driver registers:
14:29:09.222 : X 0xC0:0C:00:00
14:29:09.222 : Y 0xC0:0C:00:00
14:29:09.223 : E 0x80:0E:00:00
14:29:09.223 : Testing X connection... OK
14:29:09.331 : Testing Y connection... OK
14:29:09.331 : Testing E connection... OK
14:29:09.331 : ok

any help would be appreciated!

VOIDREP, 2019-07-03:
Interesting, even with a manual switch endstop override im only getting "Open" status, can somebody explain the need for enabling "#define SENSORLESS_HOMING"?

im assuming this is only necessary for the TMC drivers themselves? and does not influence the actual endstop configuration in any way? - i say this because when hooking up a z probe in Z-min for example no software change is necessary?

thanks!

TheNitek, 2019-07-05:
I did some further digging:
TRIGGERED state for TMC2209 survives a hardware reset
Moving the axis via GCODE sets it back to "open"

TheNitek, 2019-07-05:
Ok that's strange (CODEPENDENT_XY_HOMING off for now):
`G1 X1
ok
M119
Reporting endstop status

x_min: open
y_min: TRIGGERED
z_min: open
ok
G1 X1
ok
M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok
G1 X-1
ok
M119
Reporting endstop status
x_min: open
y_min: TRIGGERED
z_min: open
ok
G1 X-1
ok
M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok`

TheNitek, 2019-07-05:
Next observation:
I did
`G1 Y1
M119`
a few times (in relativ mode) and that's what I got:

`ok
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok
ok
Reporting endstop status
x_min: TRIGGERED
y_min: open
z_min: open
ok
ok
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok
ok
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok
ok
Reporting endstop status
x_min: open
y_min: TRIGGERED
z_min: open
ok
ok
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok
ok
Reporting endstop status
x_min: TRIGGERED
y_min: open
z_min: open
ok
ok
Reporting endstop status
x_min: open
y_min: open
z_min: open
ok
ok
Reporting endstop status
x_min: open
y_min: TRIGGERED
z_min: open
ok
`

VOIDREP, 2019-07-05:
I tried to reflash and I got USB not recognised so in the end I returned the board and now have a replacement, I'll check to see if behaviour is the same tomorrow hopefully

sn4k3, 2019-07-06:
Getting same problem, tried many things but it's always TRIGGERED. However if i return to mechanical endstops it works.

VOIDREP, 2019-07-08:
I set up a bench test today with the new SKR 1.3, going to check these, i also have a new revision of TMC2209's (FYSETC V2.1) they seem to be setup differently to my FYSETC V2.0 so ill give them a whirl as well

TheNitek, 2019-07-08:
Can you post pictures of V2 and V2.1 so I can compare?

VOIDREP, 2019-07-08:
So please forgive my potato vision camera: https://imgur.com/a/cGOylQL

the only difference i can see visually on the V2.1 is the two 0 Ohm resistors across the PDN Line jumper pads instead of the single one on the V2.0

However, looking at the IC itself they have done a much better job soldering it on the V2.1 boards i have recieved with the markings clearly visible compared to the V2.0 - i hope they didnt wash these boards lol

also i found latest Marlin 2.0 sometimes when flashed with Platform.io bricks the SKR 1.3 - had to reflash Smoothie binary included in the bigtreetech github folder back to the flash storage and recompile fresh version of marlin 2.0 one setting at a time, not sure if its an actual setting within Marlin yet. kept coming up with unrecognised USB device and i triple checked the serial settings?

VOIDREP, 2019-07-08:
ok so I think I narrowed it down to #define PIDTEMPBED, enabling this stops my SKR from working? im sure i had it enabled before though.

VOIDREP, 2019-07-08:
Ok it just gets weirder and weirder, got the V2.1 TMC2209 drivers installed and got no UART on any of them, replaced it with a known good V2.0 and got UART...so I buzzed out the 0 OHM resistors... nothing... no circuit.

solder bridged over the 0 Ohm resistor...all working fine on UART!

id inspect the jumper 0 ohm resistors on the FYSETC TMC2209 boards if your having UART issues

i would check the DIAG resistors/links but its pretty late, If my sensorless endstops still dont work thats the next thing im going to inspect.

spazwart, 2019-07-10:
I kinda have the same sort of problem.
I don't want to use stallguard yet so i have my default ender3 endstops.
The endstops only work when i remove the tmc2209 from my skr1.3
If i hold the endstop, press m119 i see triggered, if i let go it says open, all good
When i add my tmc2209 to the board, do the same thing, the endstop status allways stays open
Stallguard is disabled i guess?
m122: Stallguard thrs	0	0	0	0

teemuatlut, 2019-07-10:
Pull out the diag jumpers.

spazwart, 2019-07-10:
> Pull out the diag jumpers.

Thank you so much!

This solved all my problems, I feel so dumb!
Thanks again!!

sn4k3, 2019-07-11:
@spazwart What board you use? I have SKR1.3 + ender3 and endstops work without remove diag jumpers. While i have sensorless home ON they don't work and printer react to mechanical endstops. If i remove endstop cable it's always triggering

TheNitek, 2019-07-11:
Please create another issue if you have mechanical endstops. This issue is about sensorless homing issues with additional endstops.

bgutsev, 2019-07-12:
I am having the same issue on RAMPS 1.4. Reading 5V all the time from DIAG1 pin and when I hit Homing on X or Y the motor moves in opposite direction (normal if the home is TRIGGERED, as it is).
I have tmc2209 on X and Y and tmc2208 on Z and E0 using RAMPS 1.4 with arduino mega 2256.
I have servo motor with endstop button mount for mesh bed leveling and it is switching from open to triggered without touching anything and the servo is flickering. Can it be the signal waves or something else? I don't know...
before it was fine - Marlin 1.1.9 and tmc2208 on X and Y and drv8825 on Z and E0

BenneGehl, 2019-07-25:
I've solved the problem with tmc2209 from Fysetc on an SKR 1.3 board with sensorless homing. You only have to set one jumper under the driver board vertikal. Looking to the board from usb side you can see under the X driver 12 pins. I begin counting down left.
10  11  12
07  08  09
04  05  06
01  02  03
For adressing the tmc2209 uart at 0 set no jumper. To use uart on SKR 1.3 set the jumper vertikal between 02 and 05. Now sensorless homing is running well. Please look at the sensibility value. 255 is the highest sensibility and 0 is no sensibility. A good value for testing is 150. In this way I can stop the x and y axis by while homing. It runs! thanks to Borg on the 3d-drucker community & forum. He tested the watterott drivers but the have an other layout and the driver isn' t running with sensorless homing in my constellation. Good luck while testing.

TheNitek, 2019-07-25:
How did you solder the pads on the TMC2209? Did you have to set to jumper for sensorless homing or did your UART connection also not work without the jumper?

BenneGehl, 2019-07-25:
All the pads are  looking down. I've used the diagnotic  pin too and I've jumpered this pin to  the  endstop. If  the jumper from my last Post isn't set the uart is not running. Von meinem Samsung Galaxy Smartphone gesendet.

TheNitek, 2019-07-25:
Did you solder anything on your tmc2209? I don‘t think this solves my sensorless homing issue, since UART is working fine for me already. I just soldered the tripple pad on the bottom to enable it - no jumpers needed.

BenneGehl, 2019-07-25:
Only the diagnotic pin and the pin near him. All the pins used from skr 1.3 are connected. You Know-how I used Fysetc  driver not  Watterott.Von meinem Samsung Galaxy Smartphone gesendet.

TheNitek, 2019-07-28:
Ok, since I seem to be the only one having these problems and this issue doesn't progress anywhere I'm closing this. By now I suspect mechanical issues, so maybe this is unrelated to the TMCs.

3D-DIYde, 2019-08-03:
Just stumbled upon this thread. Too bad you've already closed it, since I do seem to have the same problem with my MKS TMC2209s on a MKS SGEN_L Board. Whatever I try, axis will only move couple mm, if any,  and trigger the endstop.
I've also disabled the sanitycheck and tried inverting X and Y endstops or disabling pullups. But nothing has worked so far.

sn4k3, 2019-08-03:
> Just stumbled upon this thread. Too bad you've already closed it, since I do seem to have the same problem with my MKS TMC2209s on a MKS SGEN_L Board. Whatever I try, axis will only move couple mm, [...]

Just like me, sensorless home doesn't work whatever i try. UART works ok and have jumper on XDIAG and YDIAG. With correct continuty from TMC DIAG to X_MIN and Y_MIN so connections are all ok

sn4k3, 2019-08-06:
I find if i set stealthchop the sensorless home works. But not in spread mode. The sensorless home only works on stealthchop ?

Christopher-Dom, 2019-08-08:
As per the TMC2209 data sheet 
[image]

sn4k3, 2019-08-08:
sanity scheck should be added or a comment near SENSORLESS_HOME to avoid similar trouble.

Christopher-Dom, 2019-08-08:
Agreed. I have set up my SKR 1.3 with Fysetc TMC 2209 v2.0. Tried running sensorless homing and it seems to have stuffed up my X_MIN signal. Can't run mechanical endstops on this anymore it always stays in one state either "TRIGGERED" or "open". Measured all voltages from the 1K, 10K, and sensor wire and the voltages change. So I can only assume that the LPC1768 is stuffed.

sn4k3, 2019-08-08:
> Can't run mechanical endstops on this anymore it always stays in one state either "TRIGGERED" or "open". Measured all voltages from the 1K, 10K, and sensor wire and the voltages change. So I can only assume that the LPC1768 is stuffed.

To use the mechanical again, you need to remove the DIAG jumper from motherboard

Christopher-Dom, 2019-08-08:
Yep removed them still no change to the state even though the voltage still toggles between 4.6V and 0.07V on the signal terminal.

sn4k3, 2019-08-08:
> Yep removed them still no change to the state even though the voltage still toggles between 4.6V and 0.07V on the signal terminal.

Looking  at datasheet you should never have 4.6V, They have a pullup from 3.3V, so should be 3.3V when open, and 0V when closed, check if wiring is correct and  good. Also check if the 1K resistor still good. Other than that damage a CPU port without affect other stuff seens hard to happen, they are also 5V tolerant

Christopher-Dom, 2019-08-08:
Thank you for the advice :)
On the sensor pin side of the 10K resistor it is 4.6V (power coming from the endstops circuit board). On the CPU side of the 10K pin is 3.3V. I will change the Pullups now and check to see if toggeling them works. From my understanding the board has 10K pullup resistors so changing the marlin code does not change anything. Is this correct?

Christopher-Dom, 2019-08-08:
Changed the pullups and still stays in the same state. Will have to buy a new board and test it. The funny thing is they worked when I had TMC2100's in.

Christopher-Dom, 2019-08-08:
I might even try resoldering the 1k resistors see if that fixes it.

sn4k3, 2019-08-08:
No, don't remove resistor.
Do basic test first.
With multimeter on beep mode put the probes on each pin of the sensor cable, press switch, it must beep = ok, otherwise switch is damaged.
If ok put in ohm mode and probe the ends of the 1k resistor. It must give you a 1k reading. If open or huge value then resistor is bad.
Do that with power off

MStorm90, 2019-08-08:
I had the same issue with my Watterott TMC2209. Their Diag pin is incompatible with the Skr 1.3. I used some Jumper wire to bridge the pin to the endstop.

Christopher-Dom, 2019-08-09:
Changed the pullups no love there, put a diode in revers on the signal so no current can flow from the endstop to the board, no love there, activated pulldown resistors no love. 
Did the beep test with switch on and off and kept one probe on the switch end of the signal wire and the other probe on the plug side of the resistors 10K, 1k, 1uF cap all beeped. 
The 1k resistor is 1k. 

Attaching the Diag to the endstop signal pin directly leaves the endstops "TRIGGERED".

Another point is that I changed the pin numbers in marlin so my x_min was set to the x_max plug and the endstops behaved normally when plugged into the x_max plug.

paulpenney, 2019-08-14:
I am having a very similar issue with the Fysetc 2209 and the Fysetc F6 board. 

1) UART is communicating with the 2209
2) DIAG is connected properly (checked with multimeter)
3) Stepper motor moves properly

Upon startup: 
M119 shows open on Y axis.

Upon Homing: 
Y doesn't move toward the min at all and bounces like it hit the endstop, shows "TRIGGERED" on M119.

Then, no matter what I do with Y after that or what sensitivity is set (0 -250), Y endstop is ALWAYS triggered and the Diag pin is putting out 5V. I have verified that it puts out 0V on startup. Looks like its stuck upon homing call?

sn4k3, 2019-08-14:
@paulpenney are you in stealchop mode on XY?

paulpenney, 2019-08-14:
Yes I have stealthchop enabled on both.

sn4k3, 2019-08-14:
@paulpenney You removed old endstops? Also DIAG pin reach X MIN and Y MIN? SENSORLESS_HOME activated on firmware?
About sensivity start with 110, verify that sensivity is set because if you just change on firmware it will not change since value is written on EEPROM once and read every time you boot. Use `M914 X110 Y110` and `M500` to fix the value

paulpenney, 2019-08-14:
Old endstops never existed - I was running 2130s with sensorless before. Diag is confirmed to reach the pins confirmed with a multimeter from the driver itself. I have tried sensitivities 20, 50, 80, 120, 150, 200, 230, 250 with the same behavior, and I have I have refreshed the eeprom and checked M914 before every test.

sn4k3, 2019-08-14:
Also you updated marlin to lastest master? 2130s require inverted logic compared with 2209

TheNitek, 2019-08-15:
I finally received my BQ TMC2209 and sensorless homing just works. So I am quite sure now this an issue with the Fysetc drivers

bgutsev, 2019-08-15:
Mine are bought from watterott. And I am using ramps. Shall I connect DIAG pin through a resistor to the endstop? And what resistor I have to use that way??

sn4k3, 2019-08-15:
> Mine are bought from watterott. And I am using ramps. Shall I connect DIAG pin through a resistor to the endstop? And what resistor I have to use that way??

You don't necessary need the resistor. But 1K is OK

sn4k3, 2019-08-15:
> I finally received my BQ TMC2209 and sensorless homing just works. So I am quite sure now this an issue with the Fysetc drivers

As you have both you can probe them. See if Fysetc 2209 ic diag pin goes to DIAG jumper

[61 more replies]

closed 2019-10-30 by boelle