MarlinFirmware/Marlin #14472: E1 Heating failed. Marlin 2.0

blueeagle69, 2019-07-01:
Since moving to my SKR 1.3 and Marlin 2.0, i'm getting a lot of E1 Heating Failed messages, with only E0 enabled.

It normally happens when the the printer is heating and I home the printer.
I've tried with #define PROBING_HEATERS_OFF enabled and disabled, but I'm still getting it.
I've attached my configuration files.
Hopefully this will help.
Thank you.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3347537/Marlin.zip)

Halestorm42, 2019-07-03:
I'm having a similar problem. I am running a SKR 1.3 and Marlin 2.0 on an Ender 3.
When I try to heat for a print with a target temp of 200 it waffles between ~188 and ~195 for a bit, then gives E1 heating failed.
Config files also attached.
Thanks!
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3352739/Marlin.zip)

thinkyhead, 2019-07-03:
I will have an SKR 1.3 for testing soon, so I can test the thermistor readings at that time.

Halestorm42, 2019-07-03:
I was actually able to solve my issue in the past couple hours. The error saying E1 heating failed is an unfixed text error according to [#12625](https://github.com/MarlinFirmware/Marlin/issues/12625).

As for the heating issue (at least my heating issue) I found [this](https://3dprinting.stackexchange.com/questions/7720/hotend-does-not-reach-set-temperature) post on stackexchange recommending calibrating PID values. I found instructions [here](https://reprap.org/wiki/PID_Tuning) and entered them into my firmware and the problem disappeared.

thinkyhead, 2019-07-03:
Marlin has always displayed "E1" to users for the first thermistor even though internally it refers to it as "E0."

closed 2019-07-03 by thinkyhead

Halestorm42, 2019-07-03:
Good to know. I’m new to Marlin (as of today, although I spent like 10 hours getting it to work) so that post was all I had to go on.