MarlinFirmware/Marlin #14474: [BUG] M503 do not show all extruders PID

mtmello2003, 2019-07-02:
If you have more than 1 extruder, M503 do not show all others PID configured by M303 or M301 gcode. 

They are stored but impossible to see the values stored...

thinkyhead, 2019-07-03:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll check them over and see if anything is amiss.

mtmello2003, 2019-07-07:
The issue happened only with the latest 2.0-bugfix.

I compiled the 25 May commit (to solve the bug fof runout sensor) and now I did the test and the second extruder PID informartion are showed with M503 with success...

boelle, 2019-09-24:
@mtmello2003 is the issue still there?

sjasonsmith, 2019-10-08:
I just encountered this issue and came here to report it. My branch is about a week old, but it doesn't look like the code has been fixed since then.

Both my extruders appear if I add the `S0` parameter to enable the "replay" printout.

Here are samples comparing the default `M503` with `M503 S0`:

### M503
```
Recv: echo:PID settings:
Recv: echo:  M301 P18.56 I1.52 D56.65
```

### M503 S0
```
Recv:   M301 E0 P18.56 I1.52 D56.65
Recv:   M301 E1 P17.65 I1.33 D58.55
```

I'm not attaching any configs, because I'm running on top of other changes which aren't buildable against bf2. I'm willing to fix this myself and submit a pull request.

The quick summary of the root cause is that the following code from configuration_store.cpp only loops over extruders if `forReplay==true`, which occurs when specifying the *S0* parameter.

```
      CONFIG_ECHO_HEADING("PID settings:");
      #if ENABLED(PIDTEMP)
        #if HOTENDS > 1
          if (forReplay) {
            HOTEND_LOOP() {
              CONFIG_ECHO_START();
              SERIAL_ECHOPAIR(
                  "  M301 E", e
                , " P", PID_PARAM(Kp, e)
                , " I", unscalePID_i(PID_PARAM(Ki, e))
                , " D", unscalePID_d(PID_PARAM(Kd, e))
              );
              #if ENABLED(PID_EXTRUSION_SCALING)
                SERIAL_ECHOPAIR(" C", PID_PARAM(Kc, e));
                if (e == 0) SERIAL_ECHOPAIR(" L", thermalManager.lpq_len);
              #endif
              SERIAL_EOL();
            }
          }
          else
        #endif // HOTENDS > 1
        // !forReplay || HOTENDS == 1
        {
          CONFIG_ECHO_START();
          SERIAL_ECHOLNPAIR(
              "  M301 P", PID_PARAM(Kp, 0) // for compatibility with hosts, only echo values for E0
            , " I", unscalePID_i(PID_PARAM(Ki, 0))
            , " D", unscalePID_d(PID_PARAM(Kd, 0))
            #if ENABLED(PID_EXTRUSION_SCALING)
              , " C", PID_PARAM(Kc, 0)
              , " L", thermalManager.lpq_len
            #endif
          );
        }
      #endif // PIDTEMP
```

thinkyhead, 2019-10-08:
Apparently this was done intentionally because some host software can only parse the `M301` output if it leaves out the `E` parameter. We should check with the @MarlinFirmware/host-software-team to see if this is still a problem. It may be that the output format issue only applies during `M303` and we can format it properly for `M503`.

sjasonsmith, 2019-10-08:
Oh interesting. I assumed this was new behavior, because I've mostly used Marlin 2.0 _after_ default M503 was inadvertently changed 8 months ago.

I see now that although this seems recent to those of us newer with Marlin 2.0. I was caught off-guard today when M503 returned "fancy" output, instead of just a list of M codes, as I had never seen it before. Personally I prefer the bluntness (and apparently completeness!) of the _forReplay_ output, so I will just make sure I use S0 from now on.

For reference, these are the commits which seem to have impacted this:

e15354e387cc825390fc4eaaf3ae4b784a8fd588 - Probably accidental change of default to "S0" (8 months ago)
dd6efe96e7154e3126cdda1a3b5afce510899bd8 - Restoration of old behavior (2 months ago)

Since it seems to be intended behavior after all, I won't be posting that pull request I mentioned in my earlier reply.

thinkyhead, 2019-10-09:
Some of the @MarlinFirmware/host-software-team is very busy right now, but I'm sure we'll get some feedback about the way they deal with the `M503` output soon.

closed 2019-10-09 by thinkyhead

thinkyhead, 2019-10-09:
I found the original change. It was done to fix compatibility with Repetier Host as it was in early 2015. https://github.com/MarlinFirmware/Marlin/pull/1344/files#diff-33b0bd580f8fc0227d430a23bb06dd5eR234

I have to assume that @repetier has addressed this now. It's unclear why it was ever parsing the output of `M503` in the first place (if indeed it was). The change was merged without getting any feedback from host developers at the time.

commit 9f4b07fb (2019-10-09): Always report all PIDs in M503

Fix #14474
commit af80653f (2019-10-09): Always report all PIDs in M503

Fix #14474