MarlinFirmware/Marlin #14476: [BUG] Running M504 causes Filament Runout distance to jump to 500mm

NovaViper, 2019-07-02:
### Description
I notice that whenever I use M504 to check my EEPROM configuration to Marlin (since I just updated my SKR V1.3 to the very latest commit of Marlin, c9be936) it sets my FILAMENT_RUNOUT_DISTANCE_MM value to 500 mm, instead of my 25mm

### Steps to Reproduce

1. Enable Filament runout and all subsequent required definitiions, ensure to have FILAMENT_RUNOUT_DISTANCE_MM defined too
2. Turn on the printer and plug into Octoprint/Pronterface
3. To get original value, first run M412
4. Run M504 to check EEPROM
5. Run M412, notice how the distance is completely different from the original we flashed

**Expected behavior:** My set settings not modified

**Actual behavior:** My settings, specifically my filament runout distance it changed

#### Additional Information

[MyMarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/3348273/Marlin.zip)
Picture of what happens:
[image]

marcio-ao, 2019-07-17:
Confirmed this bug. Will investigate.

marcio-ao, 2019-07-17:
@NovaViper: I pushed a fix for this as a PR #14661. Thank you for your bug report.

commit 839255ee (2019-07-18): Fix for M504 changing filament runout value (#14661)

Fixes #14476
closed 2019-07-18 by thinkyhead