MarlinFirmware/Marlin #14477: [FR] Individual Stealthchop activation on TMC2130 for each Axis

Massa0815, 2019-07-02:
### Description
Regarding TMC2130 in Configuration_adv.h

#define STEALTHCHOP

I want to setup Stealtchop or Spreadcycle for every axis individually.

I had some problems with losing stepps at the Z axis. Now i need to aktivate spreadcycle for Z Axis permanently.

thisiskeithb, 2019-07-02:
This is can be done with `M569` - [Set TMC stepping mode](http://marlinfw.org/docs/gcode/M569.html).

Example to disable StealthChop on Z (use spreadCycle): `M569 S0 Z`

sjasonsmith, 2020-09-11:
This is already selectable as @thisiskeithb mentioned. It seems unlikely that separating X/Y is actually useful in configuration files. That would be very strange to enable it on X but not Y, for example.

Right now this seems addressed, so I will close this.

closed 2020-09-11 by sjasonsmith