MarlinFirmware/Marlin #14478: TMC connection error on skr 1.3 with TMC2130 after using news bugfix from today

kkberg, 2019-07-02:
### Description

TMC connection error on skr 1.3 with TMC2130 after using news bugfix from today
sensorless homing does not start because of the issue

### Steps to Reproduce

1.  updating to newest configuration.h
2. creating firmware.bin

**Expected behavior:** disapearing error message from display and running 

**Actual behavior:** board starts and shows TMC connection error

#### Additional Information

[TMC_Error_Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3348821/TMC_Error_Marlin.zip)

kkberg, 2019-07-02:
Configuration worked until this morning

govido, 2019-07-02:
+1
Yesterdays version broke tmc connection for me also, I could print somehow however...
I went back to 06.29.19 bugfix.

han9988, 2019-07-03:
Had the same error on the weekend, turned out there is an updated TMCStepper Library Version 0.4.5
Do you have the current version?

thinkyhead, 2019-07-03:
Is the board powered from the 12V rail, or just USB when the problem occurs?

kkberg, 2019-07-03:
Board is powered with 24v and was already running on my printer. Went back to a version from beginning of last week which is working well. I used the complete new version. No mix of libraries.

thinkyhead, 2019-07-03:
If you are able to find the most recent commit where the error is not present, that will help us to locate the issue. If everyone did this we would be able to practically fix things while we sleep! (Which is what I'm about to do! 💤 )

kkberg, 2019-07-03:
I didn't have all the versions backwards but another user told me that his version from June 28th was the one which works and after that not. I can't verify but have an older version running and just investigating which it is.

kkberg, 2019-07-03:
My running version is from June 24th

kkberg, 2019-07-03:
Is there a chance to get access to versions between, so I could help to find out?

MercadoNegro, 2019-07-03:
Hello. I have the same issue, working on 24v.
This is new board with new stepper drivers, I thought it was a problem on the drivers, but fortunately  I saw this thread.

Thanks!
Gustavo

SHPL00P, 2019-07-03:
having the same issue as well with a brand new build, was going to start pounding my head against the wall after trying both tiny solder bridges. Can anyone have an older working version (or link) available?

govido, 2019-07-03:
Hi,
i am not really into the whole github thing and searched for an easy way to go back to specific commits.
@thinkyhead 
I think this method should be pinned somewhere to make everyones life easier if a new issue is submitted.

Go to the page where all Marlin 2.0 commits are listed:
https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x
Search for the commit number you want to try (this includes every change including that last change/commit). Copy the number/letter combination and put in this URL:
https://github.com/marlinfirmware/marlin/archive/COMMIT-SHA.zip

For example, if you want to get the last commit from July 2nd you need to put 720bc7c in the URL:
https://github.com/marlinfirmware/marlin/archive/720bc7c.zip

If you want to test June 1st just put this in: e4260a4
https://github.com/marlinfirmware/marlin/archive/e4260a4.zip

If you compare these two version, you will see that tmc2209 support was added, and many other things.
That is the easiest way i could find. Hope this helps everyone searching for older working versions.

kkberg, 2019-07-03:
Thank you, good to know

> Hi,
> i am not really into the whole github thing and searched for an easy way to go back to specific commits.
> @thinkyhead
> I think this method should be pinned somewhere to make everyones life easier [...]

Thank you, good to know.

Murray-Lindeblom, 2019-07-05:
In my case commit 0ca64a0 works.
Commit dad2f0c fails.

kkberg, 2019-07-05:
I checked [0ca64a0](https://github.com/MarlinFirmware/Marlin/commit/0ca64a0ea34f5bf96c4122b2f8341ccbcb0f6f2f) and it works for me too.

kkaleta, 2019-07-05:
> I checked [0ca64a0](https://github.com/MarlinFirmware/Marlin/commit/0ca64a0ea34f5bf96c4122b2f8341ccbcb0f6f2f) and it works for me too.

I'm try to get this working from week. Brand new SKR 1.3 and TMC2130 (whites one). 
Still: `X 0x00:00:00:00	 Bad response!/Error: All LOW`

Can you guys, build a firmware.bin for me? Simple one, with `#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3` and `#define X_DRIVER_TYPE  TMC2130`

kkberg, 2019-07-06:
> Can you guys, build a firmware.bin for me? Simple one, with `#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3` and `#define X_DRIVER_TYPE TMC2130`

Sorry, but it's off Topic here!!

kkberg, 2019-07-07:
@thinkyhead
with the latest bugfix from today, connection error is gone. Display doesn't show TMC connection error. I don't know what was changed that it's now okay. I'm going on to check this with the following versions. Thank you all

kkberg, 2019-07-07:
Sorry, but have to test again.

kkberg, 2019-07-07:
@thinkyhead 
Today I tried latest version from this morning and it's working now again!
Don't know what made it run again, but happy with it.
Thank you all.

kkberg, 2019-07-07:
TMC2130 are connecting again with Version from June 7th.

closed 2019-07-07 by kkberg

ByBack, 2019-07-08:
> TMC2130 are connecting again with Version from June 7th.

Hello Sir,
Here with a brand new setup, Fysetc white drivers (big bridge open, others 2 closed where needed) and a skr v1.3 board. 
Could you please tell me what to do with the row of red jumpers next to the endstops plugs.
And what version did you updated? Marlin bugfix or the library? If Marlin, do I need to change the whole configuration.h and configuration_adv.h ?

Thank you!

ByBack, 2019-07-08:
So quick update, sorry:
I updated my board with the newest Marlin release (7th June) and rewrote the whole config files but I am still getting the same error. Could you please send me your jumpers config and your configurations files including the one containing the pins if you made any changes to it, please?

Thank you again for your time.

kkberg, 2019-07-08:
Hello, please contact Bigtreetech support for this. I don‘t know your seiberst. Sorry

ByBack, 2019-07-08:
I do not want to bother you any longer, but my drivers are actually the same as the others regarding the pins and the board.  Could you please post your working configuration files for me to compare them? I am pretty sure to have missed something somewhere.

Thank you

kkberg, 2019-07-08:
I‘m not at at home and you get it in the evening. It for 2130 and drv8825 on E1

kkberg, 2019-07-08:
you find my config files in the first post to this thread.

ByBack, 2019-07-08:
Thank you again, appreciate it.
I will post an update if I find the answer.

ByBack, 2019-07-09:
> TMC2130 are connecting again with Version from June 7th.

Hi sir, still unsolved issue for me. Did you meant June 7th or July 7th as this post is dated from the second option? 
I tried everything I could possibly think of, rebuilt marlin with the 2 latest versions...

kkberg, 2019-07-10:
Good morning, I would  like to help you, but I need more information. Are you on Facebook? It is much easier to help with a messenger. Marlin is changing everyday and that makes it really difficult. Please give me information about your printer. Regargds

kkberg, 2019-07-10:
Which drivers on which axis?
Which mode you want to run the drivers?
Which printer (gantry, CoreXY or Delta)?
Which power 12v or 24v?
What does'nt work and what issues are occuring?
You need to provide minimum this, otherwise it's impossible to help for me! The latest versions of Marlin are running again on my Ender 3, but devil is a squirrel we say :) . If possible let's switch to Facebook, which is easier for me, because I get notified much earlier on messages.

Ozzy94, 2019-07-10:
i have a brand new build, upgraded my printer from an old ramps 1.4 to a bigtreetech skr v1.3. downloaded the marlinbugfix 2.0 yesterday from the repository.

- normal carthesian printer
- 12 Volt
- TM2130 for x,y,z and an a4986 for e0
- all pins on tmc2130 soldered on the same side (not like on the ramps where you need to bridge the spi seperately)
- placed the four jumpers on the red sockets beneath the tmc2130`s for each TMC but not for the a4986
- enabled #define TMC_USE_SW_SPI in configuration_adv.h

when i fire up the printer with the powersupply my "reprap discount FULL graphic smart controller" displays TMC connection Error. there is no sdcard placed in the cardreader of the display.

i understand that there might be a problem with my current marlin firmware version.
can someone help me

kkberg, 2019-07-10:
please post your Config

Ozzy94, 2019-07-10:
ive used these https://nl.aliexpress.com/item/32838224601.html?spm=a2g0s.9042311.0.0.44054c4dXLGQ6s the 2130 1.2 version

had them working on the ramps with spi, just soldered the spi legs back to the other side for the skr v1.3
5 volt jumper set correctly. the marlin2.0bugfix i use is not even 24 hours old

Ozzy94, 2019-07-10:
can you just explain plain and simple what you did to overcome your problem, as far as i can see my machine has the same symthomps as yours.

this thread is closed but there is no solution marked or anything

kkberg, 2019-07-10:
it was solved without doing anything. I think it depended on an Marlin internal problem which was solved with another release of th bugfix. I have not got any information from the team. But if you don't provide more information I can't help. I'm using an Ender 3 with TMC2130 an all axis and an DRV8825 on E0 and an BL-Touch. Please provide config files and I try to make a version where I know it's working.

kkberg, 2019-07-10:
[firmware.txt](https://github.com/MarlinFirmware/Marlin/files/3377207/firmware.txt)
Rename to firmware.bin and give it a try. This is my working file.  If the error is still present when booting up the printer, there will be a hardware issue may be. This is all without any warranty!!!!

gminadak, 2019-07-10:
Maybe it is completely irrelevant but i had the same problem and i found that the top row was not connected.
Check your placement and ensure that all pins are inside.
Sometimes the pins are not soldered straight and can go outside.
The next thing you can do is disable all Steppers from marlin except one (lets say X) and see if the error persist on the LCD.
And then the second one and go on till the last one.
That way you will know the driver that creates the problem.

Ozzy94, 2019-07-10:
downloaded the latest firmware, coppied my old configuration files uploaded and it runs

ByBack, 2019-07-11:
> ive used these https://nl.aliexpress.com/item/32838224601.html?spm=a2g0s.9042311.0.0.44054c4dXLGQ6s the 2130 1.2 version
>
> had them working on the ramps with spi, just soldered the spi legs back to [...]

Hi sir,
I am currently using the same drivers as yours, could you please tell me step by step how you resolved the conflict? Mine won't communicate in SPI mode whether or not I download the latest version and run it on the skr.

kkberg, 2019-07-11:
> Hi sir,
> I am currently using the same drivers as yours, could you please tell me step by step how you resolved the conflict? Mine won't communicate in SPI mode whether or not I download the latest [...]

I havn't done anything. I only used the latest Marlin bugfix at that date and it was running again. Ther was no action from my side.

ByBack, 2019-07-11:
Nevermind thanks for your help, I am tired of this issue with the spi bus.
I actually have my drivers in stand-alone stealthchop 256 interpolation and it seems to work pretty well. 

I’ll send the config files later tonight if it’s fine for you, I would just like to understand the problem.
-all jumpers on the red side (4)
-no jumpers on the diag side
-no pins soldered on the diag and vref 
-jumpers under the driver: big one on the left soldered, little one on the right soldered and little one on the left soldered.

How can I check for the length of steppers pins?

ByBack, 2019-07-11:
I also have ordered some 2208 v3.0 from bigtreetech as they seem better with the uart feature and I don’t need stallguard, got them pretty cheap for around 20 bucks / 5 pieces

kkberg, 2019-07-12:
I'm using TMC2130 V1.1 and only soldered diag1 pin for sensorless homing. Jumpers for SPI all 4 on left side. Jumpers for SPI on the right side off the board need to be set for you on XYZ. I don't know what you mean with "length of stepper pins".  I think you have jumpers soldered which are not needed for SKR 1.3 and that conflicts with the SKR.  It's really very hard to help, because config is still missing! I stop communication here, because it's to difficult for me. Please provide your Facebook name, that is much easier for me. please send phtos of all your jumper settings and drivers. I think you have to unsolder all bridges you made. But that is cristal bowl.

Ozzy94, 2019-07-12:
@ByBack did you solder the jumpers on the tmc2130, https://adrian.siemieniak.net/portal/mks-gen-1-4-board-with-tmc2130-and-marlin-firmware-howto/
Check this link for a how to

ByBack, 2019-07-12:
> @ByBack did you solder the jumpers on the tmc2130, https://adrian.siemieniak.net/portal/mks-gen-1-4-board-with-tmc2130-and-marlin-firmware-howto/
> Check this link for a how to

Yep I did!

kkberg, 2019-07-12:
where are the config files :) ?

ByBack, 2019-07-12:
> where are the config files :) ?

facebook.com/theo.glrd

[config_files.zip](https://github.com/MarlinFirmware/Marlin/files/3386433/config_files.zip)

cautioning, 2019-08-08:
I am having the same "TMC CONNECTION ERROR" with the SKR V1.3 with a build from 8/8/19.

If I use the copy of Marlin that Bigtreetech has then drivers work fine, but I can't use the printer with Octoprint.

Any suggestions?

kkberg, 2019-08-08:
> 
> 
> I am having the same "TMC CONNECTION ERROR" with the SKR V1.3 with a build from 8/8/19.
> 
> If I use the copy of Marlin that Bigtreetech has then drivers work fine, but I can't use the printer with Octoprint.
> 
> Any suggestions?

I can't build a new version in the moment. So no chance to give any feedback. Sorry

kkberg, 2019-08-08:
So, I just made a new version with today's bugfix build and it's okay. I think you have a different problem. But there is no further information that I could help you.

xotonashix, 2019-09-18:
i just get that error if i power the board over USB it seems that the Drivers doesnt get Powered over USB

ManuelMcLure, 2019-09-18:
That is correct. The board needs full power (i.e. +12V or +24V) before USB is connected (if the board is powered via USB) for the TMC drivers to be initialized correctly.

If you use the ``PS_ON`` pin on your controller board to turn on the main power supply with ``M80``, then Marlin will reinitialize the drivers after the main power supply is turned on. Otherwise, you must have 12V or 24V power before you connect USB.

summner, 2019-12-15:
> [firmware.txt](https://github.com/MarlinFirmware/Marlin/files/3377207/firmware.txt)
> Rename to firmware.bin and give it a try. This is my working file. If the error is still present when 
|Thanks for posting this file. I was able to confirm that the sticks I had actually worked (they've spent 1+year in parts bin limbo, and I had to resolder them). Now on to why my compilation is not working