MarlinFirmware/Marlin #14481: [BUG] (short description)

svsergo, 2019-07-02:
Description
I am using Marlin 2.0 with SCARA type configuration with Pronterface to send the commands. When sending the Z command home, the stepper motor runs very slowly, about 1 pulse per second. After a short time, the program ends with an accident. The X and Y move commands work correctly. Pressing Home X or Home Y individually the Z stepper motor starts running at a slow speed and everything ends with a program crash. When you download the configuration default (no SCARA), all axes are working fine.
I changed all parameters in Configuration.h according to my SCARA robot:
•	MAKERARM_SCARA
•	SCARA_LINKAGE_1, SCARA_LINKAGE_2
•	SCARA_OFFSET_X, SCARA_OFFSET_Y
•	LIMIT SWITCH PLUGS
•	ENDSTOPPULLUPS
•	DEFAULT_AXIS_STEPS_PER_UNIT
•	HOMING DIRECTION
•	Bed size
Steps to Reproduce
1.	Use SCARA (MAKERARM) configuration
2.	Try to home with limit switch configuration as X_MAX, Y_MAX
3.	Pressing Home X or Home Y individually the Z stepper motor starts running at a slow speed and everything ends with a program crash.
I tried this same configuration on 1.1 and it worked great on there.

thinkyhead, 2019-07-03:
Sorry about that. `MAKERARM_SCARA` is not supported and will soon be removed.

svsergo, 2019-07-03:
Configuration MORGAN_SCARA, the error persists. Will MORGAN_SCARA support continue?

svsergo, 2019-07-11:
Anyone have a working configuration MORGAN_SCARA?

svsergo, 2019-07-22:
In the module planner.cpp  line 2667 when is enabled  #define SCALE_FEEDRATE_SCALING not taken into account the movement in the Z-axis, so the feedrate=0. If you disable #define SCALE_FEEDRATE_SCALING, returns home along the Z-axis works. The command G28   everything ends   program crash.  Where else could there be a mistake?

boelle, 2019-09-24:
@svsergo is the issue still there?

svsergo, 2019-09-24:
Yes there is. For myself, I decided to add zero control. It may not be right, but it works.
feedrate = (HYPOT(delta[A_AXIS] - position_float[A_AXIS], delta[B_AXIS] - position_float[B_AXIS]) * duration_recip > 0) ? HYPOT(delta[A_AXIS] - position_float[A_AXIS], delta[B_AXIS] - position_float[B_AXIS]) * duration_recip : fr_mm_s;

boelle, 2019-10-21:
@svsergo is the issue still there? please test with latest 2.0.x

svsergo, 2019-10-24:
I will check after the reconstruction of the mechanism is completed.

boelle, 2019-11-03:
@svsergo is the reconstruct complte?

svsergo, 2019-11-08:
Waiting for components from China.

svsergo, 2019-11-16:
ENABLED #define SCALE_FEEDRATE_SCALING. Blocks motion  Z axis. In the new version, the error remained.

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

svsergo, 2019-12-08:
The result is the same.   In module "planner.cpp " line 2629 needs to be corrected :
const feedRate_t feedrate = (delta.z - position_float.z == 0) ?  HYPOT(delta.a - position_float.a, delta.b - position_float.b) * duration_recip : fr_mm_s ;

boelle, 2019-12-11:
so from 

const feedRate_t feedrate = HYPOT(delta.a - position_float.a, delta.b - position_float.b) * duration_recip;

to 

const feedRate_t feedrate = HYPOT(HYPOT(delta.a - position_float.a, delta.b - position_float.b), delta.z - position_float.z) * duration_recip;

?

boelle, 2019-12-11:
a PR made :-)

closed 2019-12-12 by thinkyhead