MarlinFirmware/Marlin #14483: [BUG] SKR 1.3 won't boot due to invalid EEPROM values

oneofnine, 2019-07-02:
At some point during a 1+ hour print, the SKR 1.3 eeprom settings became not what is wanted, but are apparently still valid (no CRC error it seems ?).

Due to the above, the SKR board will no longer boot due to a MAXTEMP BED error (according the LCD) if eeprom is enabled ( #define EEPROM_SETTINGS in Configuration.h). Their is no max temp error, it's a false error.

If I disable eeprom access in the config file, re-compile etc then the board boots up fine with no errors at all. If I then re-enable eeprom access in the config file, we're back to non-booting up again :(

Unfortunately, because the board won't boot with eeprom enabled, I can't re-initialize the eeprom from either the LCD menu or command line via serial.

Am stuck with a board that I can't use eeprom settings in Marlin :(

cdedwards, 2019-07-02:
wouldn't deleting the config.dat or whatever it's called from the SD card solve this issue?

ManuelMcLure, 2019-07-02:
There is no config.dat any more - the LPC1768 HAL (used by the SKR1.3) used the onboard flash to emulate EEPROM instead of a file on the SD card.

You can try checking out commit 18e396b102421fe7a82ae3d52f52a75aa44e6ed4 since that's the one before the fix that seems to have broken auto-loading of the EEPROM data on LPC1768.

ManuelMcLure, 2019-07-02:
See also #14470.

thinkyhead, 2019-07-03:
Duplicate of #14470

closed 2019-07-03 by thinkyhead

oneofnine, 2019-07-03:
> 
> 
> You can try checking out commit [18e396b](https://github.com/MarlinFirmware/Marlin/commit/18e396b102421fe7a82ae3d52f52a75aa44e6ed4) since that's the one before the fix that seems to have broken auto-loading of the EEPROM data on LPC1768.

I can confirm that is the case. That commit works, the one after does not.

Ohmarinus, 2019-11-14:
Sorry for necromancing this one, but I just had my SKR board quit on me after killing the power and starting it again. First the SD-card wouldn't show up anymore on my computer through USB. But I could still send gcode commands through Pronterface and save and retrieve new info with M500 M501 and M503.

Now when starting the printer through 12V I get this error:
Err: "MAXTEMP: Bed
paired with loud beeping.

I cannot find a way to get out of this. Can it be this is still an issue with Marlin or is this an inherent fault in the SKR 1.3's design?

OneOfEleven, 2019-11-15:
Hi @Ohmarinus 

Check your thermistor/thermistor wiring, unplug it off the board and plug a completely new one in just to first make sure that's not the problem. That one got me.

Ohmarinus, 2019-11-15:
> Hi @Ohmarinus
> 
> Check your thermistor/thermistor wiring, unplug it off the board and plug a completely new one in just to first make sure that's not the problem. That one got me.

Reflashing the firmware fixed it. I am unsure if this is an error of Marlin but clearly something got corrupted. The drawback is that I have to completely take apart the machine in order to turn off the error. It just goes stuck into bed maxtemp error.