MarlinFirmware/Marlin #14484: BIGTREETECH SKR PRO

boubouexpress, 2019-07-02:
hello

request to add the BIGTREETECH SKR PRO V1.1
ARM cortex M4 32 bit
STM32F407ZGT6 MCU

https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1

merci

govido, 2019-07-02:
I guess Bigtreetech will submit the changes needed as they advertise marlin 2.0 compatibility.
Yesterday the new E3 mini got merged from them, so I am looking forward to it.

AnHardt, 2019-07-02:
> they advertise marlin 2.0 compatibility.

That doesn't mean anything to me. Doesn't the word advertisement say all you need to know?
I remember a well know Kickstarter announcing a new Marlin compatible probe.
No other single hardware item caused that deep changes in Marlins structure to make that "compatible" crap finally work. And worst - they continue to change their product, without a warning, to make it incompatible again.

Bigtreetechs history gives me some more hope.
Let's hope for the best.

govido, 2019-07-03:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware
;)

eightheads, 2019-07-03:
> https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware
> ;)
> It is ready

Except they do not compile currently

thisiskeithb, 2019-07-04:
There’s an SKR Plus pins file in there too. 

Wonder when BigTreeTech will announce that board.

thisiskeithb, 2019-07-05:
See https://github.com/MarlinFirmware/Marlin/pull/14510, but it still doesn’t compile.

yangwenxiong, 2019-07-05:
> See #14510, but it still doesn’t compile.

Are there any errors that can be sent out that I can help solve

yangwenxiong, 2019-07-05:
@thinkyhead  Can help merge not

Msq001, 2019-07-05:
#14510 this PR  was closed because of a conflict files. I will creat a new PR for this board， and https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/tree/master/firmware this branch is work now， and Pay attention to the Readme.md

govido, 2019-07-07:
As this board is completely new I didn't want to open a new issue directly.
The things I noticed so far (my board arrived yesterday):
- TMC2208 / 2209 are not working because software uart isn't supported yet 
- graphic lcd software spi isn't working (tested with fysetc mini 2.1)
- Neopixel is not supported, the library is ignored in platformIO config for this board
- emergency parser not implemented
- lots of pin definitions are missing in the pins file (LCDs, filament runot etc.)
- auto_fan_pin does not compile
- i2c oled display is working for now
- can configure software spi for tmc2130 (not tested yet)
- no option for internal sd card on stm32? or I overlooked an option...
- serial port 0 not working

not marlin related:
- the esp 01s port is not working over usb power (not enough current?), module works when powered seperately 

If I should test something let me know.
I have tmc2130, tmc2209 to play with, tft35, fysetc mini v2.1, bltouch

Msq001, 2019-07-08:
https://github.com/MarlinFirmware/Marlin/pull/14543
1. this PR will support TMC2208/2209 uart mode
2. I have tested `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`(LCD12864) and `REPRAP_DISCOUNT_SMART_CONTROLLER`(LCD2004), They all work properly.
3. stm32 have not Serial port 0, it begin from Serial port 1
4. I have tested bltouch in `PA1` pins， it‘s ok.

I'll test your feedback one by one. Thank you very much!

govido, 2019-07-08:
@Msq001
Nice,
I will take a look again at the terminal output while compiling. I got errors regarding display and bltouch, or they are coming from somewhere else...
If stm32 has no serial 0 there should be a comment added in configuration.h ;)

govido, 2019-07-08:
@Msq001 
I tested the bltouch and fysetc options again.
I must have done something wrong in my first attempt, bltouch has no errors.
But the display throws this error right at the final compile stage...

Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\sd\usb_flashdrive\lib\message.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\sd\usb_flashdrive\lib\parsetools.cpp.o
Compiling .pioenvs\BIGTREE_SKR_PRO\src\src\sd\usb_flashdrive\usb_host.cpp.o
Linking .pioenvs\BIGTREE_SKR_PRO\firmware.elf
.pioenvs/BIGTREE_SKR_PRO/src/src/lcd/dogm/u8g_dev_uc1701_mini12864_HAL.cpp.o:(.data.u8g_dev_uc1701_mini12864_HAL_2x_sw_spi+0x8): undefined reference to `u8g_com_HAL_AVR_sw_sp_fn(_u8g_t*, unsigned char, unsigned char, void*)'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\BIGTREE_SKR_PRO\firmware.elf] Error 1

JP100099, 2019-07-09:
With the new new version over at [https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master](https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master) they seem to got the tmc2208/2209 over uart working, at least it does compile. My RepRap display stopped working though so there is probably more to do.

gerleimarci, 2019-07-16:
Please update the Neopixel library because Fysetc MINI12864 useless without it.

yellobello, 2019-09-01:
Is it currently a better idea to use bigtreetech´s sources for the SKR Pro or use the current Marlin 2.0.x-bugfix Branch? maybe dev-2.1.x?
i.e. have the sources from Bigtreetech been merged already?

thinkyhead, 2019-09-04:
> Please update the Neopixel library because Fysetc MINI12864 useless without it.

That project is at https://github.com/adafruit/Adafruit_NeoPixel

thinkyhead, 2019-09-04:
> i.e. have the sources from Bigtreetech been merged already?

They periodically push updates here.

Evg33, 2019-09-13:
https://youtu.be/uGW9yKVrxyU
https://github.com/adafruit/Adafruit_NeoPixel/pull/206#issuecomment-531297764

Evg33, 2019-09-13:
Now working with FYSETC_MINI_12864_2_1 

[image]
[image]
[image]
[image]

Evg33, 2019-09-13:
All recipe with neopixel: https://github.com/Evg33/Marlin/commit/e92b69fbf980393e914cdf37a0139ca0d892a2c6

PR for only SKR Pro pins for FYSETC Mini 12864: https://github.com/MarlinFirmware/Marlin/pull/1525

Evg33, 2019-09-15:
neopixel PR: https://github.com/MarlinFirmware/Marlin/pull/15274

Evg33, 2019-09-18:
Neopixel merged https://github.com/MarlinFirmware/Marlin/commit/12e0581f249d7aafb394a6b9ca511a9b5f337cba

Romano26, 2019-12-05:
> Now working with FYSETC_MINI_12864_2_1
> 
> [image]
> [image]
> [image]

Hello
you could send your files because I have the same configuration and I can not compile and I can not find the problem;
thank you

LinoBarreca, 2020-01-03:
@boelle close this. it's already done.

boelle, 2020-01-03:
done and thanks for the headsup

boelle, 2020-01-03:
witch such a big list one man is almost not enough :-)

closed 2020-01-03 by boelle