MarlinFirmware/Marlin #14486: when pid tuning bed, temp goes 9 to 10 degrees above intended target

shawnp30, 2019-07-02:
-->

### Description

before i use pidtune and heat the bed or hotend up the temp ramps up rather fast then overshoots intended target by 9-10 degrees. 

when using m303 e-1 U for bed pid tune i get the parameters needed but they donot work when saved with m500 using m502 and m501  show the changes, still i get over shoot when heating the bed up and hotend. the hotend when pid tuned has weird ramp up and over shoot. switched back to an older marlin 2.0 and all is fine.

[configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3352520/configuration_adv.h.txt)

[config.h.txt](https://github.com/MarlinFirmware/Marlin/files/3352518/config.h.txt)

cdedwards, 2019-07-02:
I'm seeing this as well. constantly getting thermal error's on both bed and hotend. even after I've done PID tuning on both. Temp's are not stabilizing very well, and then when you start a G29 J or G34, they bounce even more. This is on a re-arm board with 120v hotbed and 24v hotend.

thinkyhead, 2019-07-03:
We're getting a lot of reports of this in relation to the SKR board. At this point it is still a mystery.

RudyBenner, 2019-07-03:
Try the following:

**_#define BANG_MAX 230_**

govido, 2019-07-03:
Hi,
have you tried to define the pid values in the firmware and flashed it again?
I set up my pid values in the firmware and it looks like this (SKR 1.3, 24V Heaters, TMC2130):
[image]

AnHardt, 2019-07-03:
> [BUG] Bed PID tuning overshoots by 10°C

The title stillreads for me, in my words: "While i'm pid-tuning my bed i see overshoots up to 10°C." 

That's absolutely normal. The algorithm at first heats with full power on up to the target temperature. There it switches off the heater and waits until the overshoot is over and temperature has decreased to the target again. The heater is switched on again with the same power until the untershoot is over and the temperature is at the target again. Maximum-, minimum- temperatures and times are recorded. Now  a new heaterpower is calculated. Target is to center the curve at the target temperature. Circulating around the target temperature is repeated, with the new calculated power, the number of times you told in calling M303. From the last measured values the final PID-coefficients are calculated. 
So - during tuning over- and under-shoots are wanted and necessary. There absolute size does not matter much. But the more overpowered the heater is the larger the first overshoots will be. 
During the tests the shape of the curve should should converge, making it more and more symmetric to the  target temperature. Usually the overshoots decrease for that.
Here:
[image] i cant see the one or the other. At the end values are diverging again.
Without more data i can't say what is going wrong. A complete printout of this M303 could be helpful.

What did you do just before and right after tuning the bed? The shape of the graph is looking very similar. 
[image]

For the extruder tuning is looking ok.
You may want to play with `PID_FUNCTIONAL_RANGE`. Too high will wind up I more than needed. Too low will switch off full power too late. Both of the directions may increase the initial overshoot, but some are patient enough to find a sweet spot.

shawnp30, 2019-07-03:
i used pid tuning command for bed m303 e-1 s90 c8. i tried m303 e-1 U. each time bed would shoot over by 9 to 10 degrees. pids would tune and finish, i used m304 to enter pids then saved with m500 then made sure it the newest pids were available. when i save pids and start a print the bed still shoots passed the intended mark, this time being by 7-8 degrees.

DavidThijs, 2019-07-05:
Same issue here with an SKR1.1 on an E3D v6 clone. I have less problems with the bed, only the extruder.  With my other printer that is running on an MKS GEN L with an MK8 extruder, I didn't see that behavior after PID tuning. One thing that I noticed is that on my old 2560 based motherboard, the led for the mosfet is blinking a lot more when the target temperature is almost reached. I'm going to hook up a scope on the gate to see what's happening with the PWM.

AnHardt, 2019-07-05:
The frequency of blinking does not matter with that slow reacting heaters we have. The difference may be caused by hardware- vs. software-PWM. 
What matters, is only the relation of ON- to OFF-time.

boelle, 2019-09-24:
@shawnp30 is the issue still there?

shawnp30, 2019-09-24:
Yes

boelle, 2019-09-24:
@shawnp30 and this is on 2.0.x ? when did you last test?

istyszy, 2019-09-26:
on marlin 2.0.x @boelle ,
i have overshoot problems too, 
i have skr V1.3 on an anycubic delta 

marlin settings:
#define PIDTEMP
#define BANG_MAX 255     
#define PID_MAX 125
#define PID_K1 0.5      
#define PID_FUNCTIONAL_RANGE 1 

results for 230:
bias: 85 d: 39 min: 228.00 max: 234.84 Ku: 14.51 Tu: 55.42

Solo761, 2019-09-26:
A bit offtopic but it's in regard to hotend overshot. Check your thermistor, There's a chance you tugged at wires and moved/dislodged it while installing new board.

I just got SKR v1.3 board, prepared fresh Marlin, flashed it, replaced SKR v1.1 that had worked fine for last six months and lo and behold, on first print test I got wild hotend fluctuation. I set hot end to 210, it would overshoot to 218°C, then drop to 202°C, get back up to 218°C and so on... I ran PID tuning, but it didn't help, this was the best it could do after 10 cycles "bias: 98 d: 98 **min: 202.78 max: 216.67** Ku: 17.97 Tu: 49.01"

At first I thought it was this fresh marlin issue so I reinstalled old SKR v1.1 (marlin I had on it didn't have pins file for v1.3 so this was easier than to modify "old" Marlin to work on v1.3) and it now behaved absolutely the same... actually a bit worse "bias: 99 d: 99 **min: 202.22 max: 218.33** Ku: 15.65 Tu: 53.17"

In the end I tried reseating my hotend thermistor a bit and now I got this, with same marlin version as before: "bias: 55 d: 55 **min: 206.25 max: 213.12** Ku: 20.37 Tu: 41.00". Not perfect, but I think problem lies in hotend itself, not Marlin.

AnHardt, 2019-09-26:
@istyszy 
I'd start with returning to.
#define PID_K1 0.95  // 0.5 makes things much to nervous
#define PID_FUNCTIONAL_RANGE 10 // 1 means heat with full power up to 1°C below target.

istyszy, 2019-09-27:
@AnHardt i already tried PID_K1 from (1-0.5) and PID_FUNCTIONAL_RANGE (10,5,2,1)
@Solo761 you give me a good idea, i will try with new heating element / temp sensor even new hot end. (i never had these issues with anetA8 and i didn't even use PID prints came out fantastic)

Solo761, 2019-09-27:
I didn't have to change anything, just reseat thermistor. I have this printer for about year and a half and also never had this type of a problem. But in most cases issue is something ridiculous. 

Just a small update to my previous post, values that are written in "bias" line aren't representative of achieved precision. It's just what Marlin used as low and high when calibrating. After 10 cycles I got to min 206 and max 213 degrees Celsius, but when I set my hotend to 210°C after that regularly it got to 210 and held it, occasionally dropping it to 209 and quickly back to 210. 
So above values when doing PID tune are actually OK :).

AnHardt, 2019-09-27:
Autotune does not try to minimize the difference between these min and max. It tries to adjust the heater-power to get a symmetrical wave around the set point. From that power, the amplitude of the wave and the wave-length it calculates the heater properties Ku and Tu and form them it tries to estimate useful PID parameters. 
Getting high amplitudes during Autotuning just means, the heater has a high thermal capacity and/or  delay from production of the heat, by the heat element, to measuring the altered temperature, by the sensor.

boelle, 2019-10-21:
i cant confirm this one, there is no overshot at all
```
[...]
Send: M303 E-1 S60 C8
Recv: PID Autotune start
[...]
Recv:  bias: 57 d: 57 min: 59.61 max: 61.29 T:30.34 /0.00 B:59.87 /0.00 C:26.82 /0.00 @:0 B@:57
[...]
Recv:  bias: 59 d: 59 min: 59.41 max: 60.16 T:30.65 /0.00 B:60.00 /0.00 C:26.90 /0.00 @:0 B@:59
[...]
Recv:  bias: 78 d: 78 min: 58.80 max: 60.00 Ku: 165.39 Tu: 15.38
Recv:  No overshoot
Recv:  Kp: 33.08 Ki: 4.30 Kd: 169.52
[...]
Recv:  bias: 50 d: 50 min: 58.96 max: 61.13 Ku: 58.66 Tu: 60.86
Recv:  No overshoot
Recv:  Kp: 11.73 Ki: 0.39 Kd: 237.99
[...]
Recv:  bias: 50 d: 50 min: 59.41 max: 60.13 Ku: 177.01 Tu: 10.25
Recv:  No overshoot
Recv:  Kp: 35.40 Ki: 6.91 Kd: 120.96
[...]
Recv:  bias: 74 d: 74 min: 59.41 max: 60.05 Ku: 295.06 Tu: 20.02
Recv:  No overshoot
Recv:  Kp: 59.01 Ki: 5.90 Kd: 393.81
[...]
Recv:  bias: 47 d: 47 min: 58.95 max: 60.48 Ku: 77.83 Tu: 41.51
Recv:  No overshoot
Recv:  Kp: 15.57 Ki: 0.75 Kd: 215.37
[...]
Recv:  bias: 47 d: 47 min: 59.41 max: 60.02 Ku: 196.72 Tu: 10.22
Recv:  No overshoot
Recv:  Kp: 39.34 Ki: 7.70 Kd: 134.07
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
Recv: #define DEFAULT_bedKp 39.34
Recv: #define DEFAULT_bedKi 7.70
Recv: #define DEFAULT_bedKd 134.07
Recv: ok
```

[image]

given the lack of activity i will close this one
we can alway reopen if there is a confirmed issue with marlin itself

closed 2019-10-21 by boelle