MarlinFirmware/Marlin #14489: [BUG] (Bugfix 2.0) Homing Failed Printer halted

Bergerac56, 2019-07-03:
With the last bugfix : 6a2a10a154c9db39d5b3732e66685953e5f945e7 and when issuing a G28 or using the LCD option to home the printer, I get a Homing Failed / PRINTER HALTED. On pronterface, I get: 
SENDING:G28
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

The printer is unusable.

The previous version I used : a0f8b25c1bd6826aa44edca4812a597dc433a13a does not have this issue. I did not have yet the time to try to narrow down to the commit which could be the problem.

Did I do something wrong?

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3356182/config.zip)

Glod76, 2019-07-03:
same problem mks sbase
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3356856/Marlin.zip)

Glod76, 2019-07-03:
also bed doesn't heat Bed heating failed printer halted please reset

Glod76, 2019-07-04:
any update for a fix to this?

runningz, 2019-07-04:
Me too. But i can't do anything (homing, move, heat bed....) :-( I think it's a me fault.... but i see i'm not alone.

nebman, 2019-07-04:
same problem with SKR 1.3

Bergerac56, 2019-07-04:
Still looking at which commit provoked it

kihope, 2019-07-04:
same for me with skr 1.3 it is in soiurce code and propably doesnt matter what board is used

kihope, 2019-07-04:
I tracked the commit

commit: 108d0df
link: https://github.com/MarlinFirmware/Marlin/commit/108d0df56532e4eb30467d36b0fb2aa5bf4f50be

after this commit the problem started i tested it by "build and test"

robbycandra, 2019-07-04:
Already reverted with de462f2b

kihope, 2019-07-05:
Then i rly dont know, version before commit i mentioned works with and after does not

Bergerac56, 2019-07-05:
Just tried with the last bugfix b6546ea33a0f6eebee520dda516e04d3b68ded55 and it does still not work. Even if everything is possible, I doubt that a commit affecting SD EEPROM emulation could provoque this issue... But you never know

kihope, 2019-07-05:
There is still posibillity i am not right, however commit under works, also last commit i tryied before the one i mentioned were like 3 up and it want working so.. i dont know now i use older version what i dont mind at all

nebman, 2019-07-05:
b6546ea33a0f6eebee520dda516e04d3b68ded55 doesnt compile because of broken macros but if I omit only this specific commit, the homing problem went away

kihope, 2019-07-05:
cool

Bergerac56, 2019-07-06:
The last version: 8916acf0831a39462bbbc0a3e5ae570648199135 (at least) solves the issue for me. No compiling and no homing problems anymore.

closed 2019-07-06 by Bergerac56

Mbombace, 2019-07-07:
Same problem on SKR v1.3 with TMC drivers. Does the newest version fix this? 

Is there an easy way to update firmware without having to re-enter all of my firmware settings?

yet-another-average-joe, 2019-07-08:
@Mbombace

> Is there an easy way to update firmware without having to re-enter all of my firmware settings?

Marlin-Configurator by @akaJes does it nearly auto-magically with a simple drag/drop operation. It will forget some parameters (on my side GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y, so I have to verify with a file comp software (under Windows : WinMerge).

https://github.com/MarlinFirmware/Marlin/issues/6722

https://github.com/akaJes/marlin-config

Works for Configuration.h and Configuration_adv.h only. Other files (pins for example) have to be edited by hand). But after some practice, Marlin-Configurator allows the user to upgrade/compile/flash a new commit within less than 5 mn.

I don't know of another app doing this.

nebman, 2019-07-12:
> Is there an easy way to update firmware without having to re-enter all of my firmware settings?

Use GIT: create a local branch, commit your setting changes to it and whenever you want to update your firmware, merge the 2.0-bugfix branch into your local branch and let git do the merging magic :D The Github Desktop App is pretty easy to use in this case

chrashing, 2020-06-14:
I just fixed a 'Printer halted. kill() called!'  problem while homing after  upgrading  my ender 3 pro  to a skr v1.4 board.  For me, turned out to be the stepper motor driver signals coupling to the endstop switch.  The fix for me was to move the wires for stepper motor away from the 2 switch wires in the ribbon cables, and run them separately.    Saw the problem on both the X and Y axis. The Z axis already had the wires separated.  I expect the new stronger motor TMC2209  driver signals increase the coupling noise, or maybe the skr switch sensors are more sensitive to noise.   This problem was exacerbated for my mahine having longer ribbon cables than stock.

chrashing, 2020-06-15:
Later found that In the marlin configuration.h file there is a  ENDSTOP_NOISE_THRESHOLD setting that I probably could have used instead of separating the endstop switch wiring from the stepper motor wiring. I have not tried that yet.