MarlinFirmware/Marlin #14490: [BUG] G29 UBL Kompare not working

RobReus, 2019-07-03:
### Description

Using `G29` with the `K<index>` parameter does not work. Nothing is subtracted from the current mesh.

### Steps to Reproduce

1. Bed level using G29 as normal and save to slot 1
2. Repeat step 1 but save in slot 2
3. View current mesh using `G29 T0`
4. With slot 2 as active mesh, issue `G29 K1`
5. View current mesh again using `G29 T0`

**Expected behavior:**
Upon issuing `G29 K1` on mesh slot 2, I except the mesh of slot 1 to be subtracted from the mesh in slot 2 and this to be visible when executing `G29 T0`.

**Actual behavior:**
The mesh in slot 2 is 100% the same after the diff (`G29 K1`) as it was before. This should not be the case.

#### Additional Information

Configuration.h and Configuration_adv.h: [configs.zip](https://github.com/MarlinFirmware/Marlin/files/3356206/configs.zip)

Printer is running on commit sha `720bc7c00b0eedf2e7601d5fc6e6ff61f9833562` from `bugfix-2.0.x`.

```Send: G29 L1
Recv: Mesh loaded from slot 1
Recv: Done.
Recv: ok
[...]
Send: G29 T0
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,200)                      (240,200)
Recv:         0       1       2       3       4
Recv:  4 | +0.395  +0.282  +0.159  +0.049  -0.054
Recv:    |
Recv:  3 | +0.367  +0.232   0.000  +0.013  -0.100
Recv:    |
Recv:  2 | +0.364  +0.118 [-0.001] -0.076  -0.086
Recv:    |
Recv:  1 | +0.395  +0.166  +0.010  +0.036  -0.011
Recv:    |
Recv:  0 | +0.522  +0.297  +0.195  +0.171  +0.233
Recv:         0       1       2       3       4
Recv:     ( 10, 10)                      (240, 10)
Recv: 
Recv: ok
[...]
Send: G29 L2
Recv: Mesh loaded from slot 2
Recv: Done.
Recv: ok
[...]
Send: G29 T0
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,200)                      (240,200)
Recv:         0       1       2       3       4
Recv:  4 | +0.424  +0.290  +0.174  +0.061  -0.052
Recv:    |
Recv:  3 | +0.390  +0.240  +0.020  +0.040  -0.074
Recv:    |
Recv:  2 | +0.424  +0.138 [+0.012] -0.049  -0.059
Recv:    |
Recv:  1 | +0.396  +0.193  +0.008  +0.035  +0.008
Recv:    |
Recv:  0 | +0.497  +0.321  +0.205  +0.189  +0.242
Recv:         0       1       2       3       4
Recv:     ( 10, 10)                      (240, 10)
Recv: 
Recv: ok
[...]
Send: G29 K1
Recv: ok
[...]
Send: G29 T0
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,200)                      (240,200)
Recv:         0       1       2       3       4
Recv:  4 | +0.424  +0.290  +0.174  +0.061  -0.052
Recv:    |
Recv:  3 | +0.390  +0.240  +0.020  +0.040  -0.074
Recv:    |
Recv:  2 | +0.424  +0.138 [+0.012] -0.049  -0.059
Recv:    |
Recv:  1 | +0.396  +0.193  +0.008  +0.035  +0.008
Recv:    |
Recv:  0 | +0.497  +0.321  +0.205  +0.189  +0.242
Recv:         0       1       2       3       4
Recv:     ( 10, 10)                      (240, 10)```

Roxy-3D, 2019-07-03:
From what you posted, I understand why the issue is was opened.   But I did a little bit more comprehensive test, and everything seems to work OK.   Can you do the steps I did?

I did:
- G29 L1             (load mesh 1)
- G29 P5             (get statistics on mesh 1)
- G29 P0             (to create a zero'd out mesh)
- G29 P6 C ?.?????     (Use what ever Mean Mesh Height value was just reported to you with the G29 P5)
- G29 T               (verify the entire mesh has the same Mean Mesh Height value in it)
- G29 S4             (save the mesh in slot 4)
- G29 L1             (load mesh 1)
- G29 K 4            (subtract off mesh 4)
- G29 T P5          (view altered mesh and its statistics)

If the code is working correctly, the last 'G29 T P5' should show a Mean Mesh Height of 0.0000 and all of your mesh values should be altered by the size you selected on the 'G29 P6 C'

Roxy-3D, 2019-07-04:
Oh!   I see the problem....   Check out your commands:

Recv: Mesh loaded from slot 1
Recv: Done.
Recv: ok
[...]
Send: G29 T0

The 'T' command only displays the current mesh.   It doesn't go look inside the EEPROM slots at which ever slot you specify.   The 'T' command only allows the parameter (if it is given) to specify the type of display to do.      

Here is the comment section for the 'T' command (in the source code).    It doesn't offer the ability to specify a parameter:
```
   *
   *   T     Topology   Display the Mesh Map Topology.
   *                    'T' can be used alone (e.g., G29 T) or in combination with most of the other commands.
   *                    This option works with all Phase commands (e.g., G29 P4 R 5 T X 50 Y100 C -.1 O)
   *                    This parameter can also specify a Map Type. T0 (the default) is user-readable. T1 can
   *                    is suitable to paste into a spreadsheet for a 3D graph of the mesh.
   *
```

Roxy-3D, 2019-07-05:
@RobReus    Do you agree?    Can we close this?

RobReus, 2019-07-05:
@Roxy-3D sorry for the late reply. I have not yet been able to this the above myself, will do it this weekend. Will let you guys know if it is solved for me.

On a side note, I remember that in the past, it did show the subtract when doing `G29 T`, is this behavior changed in the past or is my memory totally failing me?

RobReus, 2019-07-06:
@Roxy-3D 
I follow the exact same steps you posted and this is the result:
```Send: G29 L1
Recv: Mesh loaded from slot 1
Recv: Done.
Recv: ok
[...]
Send: G29 P5
Recv: # of samples: 25
Recv: Mean Mesh Height: 0.148220
Recv: Standard Deviation: 0.163669
Recv: ok
[...]
Send: G29 P0
Recv: Mesh zeroed.
Recv: ok
[...]
Send: G29 P6 C 0.148220
Recv: ok
[...]
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,200)                      (240,200)
Recv:         0       1       2       3       4
Recv:  4 | +0.148  +0.148  +0.148  +0.148  +0.148
Recv:    |
Recv:  3 | +0.148  +0.148  +0.148  +0.148  +0.148
Recv:    |
Recv:  2 | +0.148  +0.148  +0.148  +0.148  +0.148
Recv:    |
Recv:  1 | +0.148  +0.148  +0.148  +0.148  +0.148
Recv:    |
Recv:  0 |[+0.148] +0.148  +0.148  +0.148  +0.148
Recv:         0       1       2       3       4
Recv:     ( 10, 10)                      (240, 10)
Recv: 
Recv: ok
[...]
Send: G29 S4
Recv: Mesh saved in slot 4
Recv: Done.
Recv: ok
[...]
Send: G29 L1
Recv: Mesh loaded from slot 1
Recv: Done.
Recv: ok
[...]
Send: G29 K 4
Recv: ok
[...]
Send: G29 T P5
Recv: # of samples: 25
Recv: Mean Mesh Height: 0.148220
Recv: Standard Deviation: 0.163669
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,200)                      (240,200)
Recv:         0       1       2       3       4
Recv:  4 | +0.419  +0.265  +0.155  +0.045  -0.064
Recv:    |
Recv:  3 | +0.373  +0.219  +0.015  +0.023  -0.083
Recv:    |
Recv:  2 | +0.372  +0.145  +0.013  -0.057  -0.069
Recv:    |
Recv:  1 | +0.416  +0.183  +0.002  +0.021  -0.005
Recv:    |
Recv:  0 |[+0.456] +0.282  +0.191  +0.170  +0.220
Recv:         0       1       2       3       4
Recv:     ( 10, 10)                      (240, 10)
Recv: 
Recv: ok```

It looks like the last `G29 T P5` does not show a mean mesh height of 0.0000. Does that mean I am running into a bug or am I doing something wrong?

Roxy-3D, 2019-07-14:
Does it show it a mean mesh height of +/- .001  ?   If so... close enough for government work.

boelle, 2019-09-24:
@RobReus is the issue still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle