MarlinFirmware/Marlin #14494: [CASE_LIGHT_MENU] if enabled fails to compile on SKR1.3

GizmoTheGreen, 2019-07-03:
### Description

Fails to compile with case light menu option

### Steps to Reproduce
I have edited pins like so:
#if HOTENDS == 1
  #define FAN1_PIN         P2_04
#else
  #define CASE_LIGHT_PIN     P2_04
#endif

Enabling only #define CASE_LIGHT_ENABLE compiles just fine.
however enabling #define CASE_LIGHT_MENU gives the error below

#### Additional Information

Linking .pioenvs\LPC1768\firmware.elf
c:/users/myuser/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: .pioenvs/LPC1768/src/src/lcd/menu/menu_configuration.cpp.o: in funct
ion `menu_case_light()':
E:\Userdata\Desktop\Marlins\Marlin-bugfix-2.0.x SKR+fysetc12864/Marlin\src\lcd\menu/menu_configuration.cpp:245: undefined reference to `case_light_brightness'
c:/users/myuser/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: E:\Userdata\Desktop\Marlins\Marlin-bugfix-2.0.x SKR+fysetc12864/Marl
in\src\lcd\menu/menu_configuration.cpp:245: undefined reference to `case_light_on'
c:/users/myuser/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: E:\Userdata\Desktop\Marlins\Marlin-bugfix-2.0.x SKR+fysetc12864/Marl

in\src\lcd\menu/menu_configuration.cpp:245: undefined reference to `update_case_light()'

Ludy87, 2019-07-04:
I can not adjust the error, please add the config files and pin _ *. h file with.

GizmoTheGreen, 2019-07-04:
[Marlinconf.zip](https://github.com/MarlinFirmware/Marlin/files/3360495/Marlinconf.zip)
here try this, note I downloaded 2.0 bugfix 07-03, 18:43. if yours is older or newer maybe it's fixed I don't know... just got my skr 1.3 yesterday.

Ludy87, 2019-07-05:
`#define CASE_LIGHT_PIN P2_04` is never defined, because` #if HOTENDS == 1` `return true`.

My suggestion (not checked):
```c++
#if HOTENDS == 1
  #define FAN1_PIN -1
#endif

#define CASE_LIGHT_PIN P2_04
```

GizmoTheGreen, 2019-07-05:
wow. then it's my mistake. thanks.

closed 2019-07-05 by GizmoTheGreen