MarlinFirmware/Marlin #14496: snippet of HAL code borks the fan output of SKR Mini V1.1

peterkrull, 2019-07-04:
### Description

I'm not exactly a shark at programming, so troubleshooting this took me way too long. I also don't know what the permanent fix looks like, so I guess I will report it as a bug.

With a current build of Marlin Bugfix 2.0.x tailored for the SKR Mini V1.1 the fan output on the normal FAN/FAN0 output always runs at roughly 60% power, no matter what I set the fan speed to on the LCD. I tried rolling back about 20 days to when Bigtreetech built their own (not maintained) version of Marlin, and there it worked.

I narrowed it down to this commit:
https://github.com/MarlinFirmware/Marlin/commit/52383633e7062bf5ffd72d9c41c4cf7df05c544c

_and more specifically in Marlin/src/HAL/HAL_STM32F1/HAL.h_ 
void analogWrite(pin_t pin, int pwm_val8); // PWM only! mul by 257 in maple!?

_and in Marlin/src/HAL/HAL_STM32F1/HAL.cpp_
// Wrapper to maple unprotected analogWrite
void analogWrite(pin_t pin, int pwm_val8) {
  if (PWM_PIN(pin) && IS_OUTPUT(pin))
    analogWrite(uint8_t(pin), pwm_val8);
}

Commenting out the line in HAL.h makes the fan work as expected

As I said, I don't exactly know what the bit of code does, and I don't just want to commit my "fix" of commenting it out or deleting the code just yet. 

@tpruvot and
@thinkyhead 

### Steps to Reproduce

1. Build the latest version of Marlin Bugfix 2.0.x for the SKR Mini V1.1
2. The fan is now **not** controllable

1. Do the exact same thing, but comment out //void analogWrite(pin_t pin, int pwm_val8);
2. The fan is now controllable

tpruvot, 2019-07-04:
PC8 seems associated to timer8, channel 3... Could you try to remove "&& IS_OUTPUT(pin)" in the hal ?

analogWrite use on a non-pwm pin will just crash marlin without this function... but IS_OUTPUT() check is maybe not required.

else FAN_SOFT_PWM can be used, which allow also to tune the min/max speed

peterkrull, 2019-07-04:
Removing "&& IS_OUTPUT(pin)" seems to do the trick actually.

If a PWM_PIN is always an output anyway, there is no harm in just keeping the "&& IS_OUTPUT(pin)" part out, right?

tpruvot, 2019-07-04:
in theory yep, in practice there can be normal inputs on timer associated pins (its what PWM_PIN() checks)

closed 2019-07-11 by thinkyhead