MarlinFirmware/Marlin #14497: [FR] Sensorless filament run out

joseitor, 2019-07-04:
Hi community!

It have just came to my mind that if with the new TMC2209 we can use Trinamic in the extruder easyly and with StallGuard4, maybe it would be possible to measure the force that the extruder motor is applying and when it drop below a value, we can detect that the filament run out and stop the printing as the same way that with the filament sensor.

Do you thing this is possible? I couldn't test the TMC yet, and also I don't know if it's sending the force feedback all the time or it's necessary to make a special movement to measure it in witch case this wouldn't be possible.

It's just a idea, sorry if it's not possible yet.

mylife4aiurr, 2019-07-05:
I love this idea..... detecting filament runout, but possibly even detecting a jam also.

yet-another-average-joe, 2019-07-06:
Interesting !

But the filament is generally hooked to the spool, and sometimes it is even glued to... I've never seen anything else, and never understood the interest for the end of filament detectors that can be seen everywhere (always tested with cutting pliers...). Therefore, an end of filament situation can be compared to a tangle. And if it pops off the spool, the hook will get stuck at the extruder input. The force will not drop until the gear heavily grinds the filament.

Moniroring the force could also detect a clogged nozzle... But in any case (end of filament, tangle, jam, clog), the reference (normal force) depends on the nozzle diameter and extrusion speed (it's easier to push through a 1.2mm nozzle than through a 0.2mm one).
It also depends on the material viscosity and the extruder gear preload (slipage, grinding).
It could be tested with an Arduino and the TMCStepper library (https://github.com/teemuatlut/TMCStepper)
An example for StallGard comes with the library (for a proof of concept).

marcio-ao, 2019-07-24:
At one point I performed some experimentation with this and appeared that stallguard wasn't sensitive enough to detect a filament stall. Then again, our extruders are geared, so possibly that is a factor. Not saying someone can't figure it out, but just saying that it may not be that easy :)

marcio-ao, 2019-07-24:
> I don't know if it's sending the force feedback all the time or it's necessary to make a special movement to measure it in witch case this wouldn't be possible.

It's can be read anytime the motor is in motion, so you can get readings while extruding. The question, however, is whether you will get meaningful values. This is where things get a bit dicey :)

7eggert, 2019-08-03:
I have a direct drive extruder. It does not stall on jams but grinds the filament.

sustmi, 2020-03-29:
Today I installed Marlin 2.0.5.2 that [supports reading `SG_RESULT` value using `M122` command](https://github.com/MarlinFirmware/Marlin/blob/2.0.5.2/Marlin/src/feature/tmc_util.cpp#L636).
I found out that the `SG_RESULT` value for the extruder stepper motor during print was almost always 2 or 0 (like at stall). Except for fast retraction moves when I managed to catch values like 166.

I guess that extruder movement speed is too low for StallGuard4 (I have TMC2209) to work properly.

The [TMC2209 datasheet](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf) states:
> StallGuard4 does not operate reliably at extreme motor velocities: Very low motor velocities (for many motors,  **less  than  one  revolution  per second**) generate a low back EMF and make the  measurement unstable and dependent on environment conditions (temperature, etc.)

According to the [print speed calculator](https://www.printitindustries.com/pages/print-speed-calculator) for:
- extrusion width: 0.48 mm
- layer height: 0.2 mm
- print speed: 100 mm/s (which is quite high)
- filament diameter: 1.75 mm

the extruder speed is: 3.99 mm/s.
With my original extruder on Creality Ender 3 which has extruder cogwheel with ~10.5mm diameter  that means ~7.25 rpm or **0.12 revolutions per second**.

Both the theory and the practice show that extruder motor moves too slow for StallGuard to be usable.

Maybe if there was an aggressive gear ratio (more than 10:1) on the extruder motor it could be usable. However I am not sure if the resolution would be sufficient then.
Also, such aggressive filament ratio would make the extruder grind the filament instead of motor skipping step when the filament gets jammed. 

For reference:
My motor is the original stepper motor on Creality Ender 3 ([JK42HS40-1004AC](https://ecksteinimg.de/Datasheet/Schrittmotor/JK42HS40-1004A/JK42HS40-1004AC.pdf)), the driver voltage is 24V and the RMS driver current for extruder stepper is 642 mA (the value set in the printer menu).

italocjs, 2020-06-07:
I have experimented with this, and didnt had success. there were just too many variables for bowden (as distance, fricction, type of filament (some are naturally harder to push) and spring load.  I'm experimenting now using a simple encoder, which is showing really good results (i can detect filament runout, full and partial clog based on expected / real movement performed).  Im just using the same encoder that some screens use (<0.5 USD) and a microswitch (<0.1 USD)

DrumClock, 2020-12-22:
Hi,
how I test this sensor and it is very reliable .. https://github.com/MarlinFirmware/Marlin/issues/19214
It is built on arduino NANO but would like to integrate it into Marlin .... but no one is interested in it ....