MarlinFirmware/Marlin #14501: Ender 3 Pro - Enabling BLTOUCH_V3 causes early pin retraction for BLTouch V3.1

OtiNan, 2019-07-04:
### Description

Enabling BLTOUCH_V3 in Configuration.h switched probing mode "Probing with Touch Switch Mode" ( M280 P0 S60) which in turn causes early pin retraction while doing a G29 and results in erroneous bed mesh.   

This happens about 50% of the time and it happens regardless if you remove the capacitor on the z endstop of the Ender 3 mainboard.

### Steps to Reproduce

1. Install BLTouch V3.1
2. Enable BLTOUCH and BLTOUCH_V3 in Configuration.h and flash the firmware
3. Do a G29
4. Observe early pin retraction and erroneous bed mesh

**Expected behavior:** [What you expect to happen]
Complete a G29 succesfully

#### Additional Information

Removing the capacitor from the z endstop of the Ender 3 mainboard and commenting out the BLTOUCH_V3 define switches to "Probing with Push-pin Down" solves the issue.

From reading the code and with my limited understanding, if BLTOUCH_V3 is defined the set_bltouch_deployed procedure sends a M280 P0 S60 (Touch Switch Mode) after ever deploy (M280 P0 S10) or a stow (M280 P0 S90). Commenting that out or disabling BLTOUCH_V3 resolves the issue of early pin retraction.

Is the only reason of using Touch Switch Mode probing with BLTouch >=V3 the capacitor or am i missing something?

tpruvot, 2019-07-04:
> Marlin/src/inc/SanityCheck.h:214:#elif defined(BLTOUCH_V3)
> Marlin/src/inc/SanityCheck.h:215:  #error "BLTOUCH_V3 is obsolete. Please update your configuration."

OtiNan, 2019-07-04:
That is for 2.0. The changes for BLTouch V3 were moved to bugfix-1.1.x ([1.1.x] BLTOUCH_V3 parity with 2.0.x (#13682)) and that is still there.
Should i just move to 2.0 and consider this closed?

closed 2019-07-05 by OtiNan