MarlinFirmware/Marlin #14507: [BUG] Anet A8 profile x/y offsets transposed

rigelhope, 2019-07-04:
In the sample configuration for the anet a8, X_MIN_POS and Y_MIN_POS offsets are transposed.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RudyBenner, 2019-07-04:
Works for me.

[image]

RudyBenner, 2019-07-04:
Since you did not provide any information, you will get nothing more.

RudyBenner, 2019-07-04:
without supplied the required config files, you cannot expect more.

rigelhope, 2019-07-05:
it's more of a nudge, so that people searching while aggravated because Y IT NOT WORK?!?! will get prompted to review those constants during the course of their troubleshooting.

the y-axis is, per the build/setup instructions, the stepper that drives the belt on the gantry, and it has the bigger offset between the x and y axes (i wound up adjusting it to -43 for my own purposes, and made the bed size smaller), so -33 is clearly the y offset, not the x. if you've transposed cables or something it will work, of course.

since the general recommendations with the Anet a8 are pretty universally to install marlin immediately so as to mitigate fire risk, some users may be unprepared for going over the configuration files with a fine toothed comb.

ManuelMcLure, 2019-07-05:
Then the instructions are wrong. The Y axis is the one that moves the bed back and forth. The X axis is the one on the gantry. This was the case with my Anet A8 and is the case for all i3-style machines.

ManuelMcLure, 2019-07-05:
Wait - is this an A8 (acrylic frame) or A8 Plus (aluminum extrusion frame)? These are very different printers even though they're both identified as "A8".

ManuelMcLure, 2019-07-05:
https://www.china-gadgets.de/app/uploads/2016/11/A8-3D-Printer-Installation-Instructions-2016-6-30.pdf - shows the X motor on the gantry, not the Y motor.

Patag, 2019-07-05:
@rigelhope 
I've been using an A8 (acrylic frame) for 6 months or so and I can confirm the X/Y_MIN_POS are correct (X=-33, Y=-10). Maybe your endstops are not fitted IAW instructions or you are talking about A8+ or you swapped X/Y wiring to steppers

rigelhope, 2019-07-05:
yep, acrylic frame. reviewed instructions, i had them transposed. not sure how it happened but happened it did. closed.

closed 2019-07-05 by rigelhope

Patag, 2019-07-05:
Saying "Thanks" won't burn your mouth, knowing it's not a support site here.