MarlinFirmware/Marlin #14508: Marlin G33 How does it work

nano1900, 2019-07-05:
Hi,i just want to know ,how does the G33 work.there is any documents

Chalmes, 2019-07-05:
Does this help? (http://marlinfw.org/docs/gcode/G033.html)

nano1900, 2019-07-05:
thank you Chalmes 
I want to konw the theory behind G33

AnHardt, 2019-07-05:
For the details you have to look into the code.

In brief and simplified:
A delta calculates from the cartesian xyz position and some parameters describing the geometry, the needed carriage positions on the abc-towers.
Measuring the *exact* geometry is difficult (for humans) at the level of exactness we want.
If one or more of these geometric parameters is different from reality, a plane and level surface (bed) appears distorted - when trying to move *on* the surface the nozzle will dig into the bed at places and be above the bed at others.
So G33 measures (probes) the distortion of the bed, compares the values to a ideally plane and level surface and tries to find a better geometry from the distortion.
With the assumption of the new geometry the probe-calculate procedure is repeated.
The more points are probed, the more geometric parameters can be corrected.

nano1900, 2019-07-05:
thank you AnHardt . i think ,i'm get it a littel bit.
It's just like a optimization algorithm like Gradient Descent and the more iteration step you take more accurate parameters you will get.
Am i right ?

thinkyhead, 2019-07-11:
If you like discussing this kind of thing, you might want to join at https://discord.gg/n5NJ59y

closed 2019-07-11 by thinkyhead