MarlinFirmware/Marlin #14511: [FR] (HAL for 8051)

damo2641, 2019-07-05:
Hi community!

I want to find some experts to development HAL for our 8051 chip-T5L.  
Our chip is mainly used in our major product :TFT LCD UART modules
[image]
It has already applied in many Chinese 3D printer manufacturer for it's cost performance and convenience for deveopment.
But our new generation chip T5L could be a single chip solution for 3D printer. Because it is a dual core processor. The GUI core is used for TFT and touch driver and the OS core is a 8051 core.   
The only barrier before us is Marlin. So we really need the HAL for 8051.  Here is the development guide for our T5L LCD mpdules.
[DGUSII_T5L Application Development Guide20190528V16.docx](https://github.com/MarlinFirmware/Marlin/files/3361369/DGUSII_T5L.Application.Development.Guide20190528V16.docx)
My email: eric@dwin.com.cn

thinkyhead, 2019-07-11:
Serial controllers are implemented through the `ExtUI` API in Marlin. There are many talented programmers around the world who should be able to implement this for you. We recommend you also post your job offer in the following public forums, and be sure to state how much you are willing to pay.

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

damo2641, 2019-07-12:
Thanks for your advice!

Sincerely
Mohan Chen

Eric 
BEIJING DWIN TECHNOLOGY CO.,LTD.
18F, Building 5#, Jia 2#, Xi San Huan Bei Rd, Haidian Dist, 100081, Beijing, China
CELL: +86.18611305191 Skype:Mohan Chen
E-MAIL: Eric@dwin.com.cn

2019-07-12 

-------陈漠瀚--------
北京迪文科技有限公司
网站：www.dwin.com.cn
电话：18611305191 传真：010-82157883（北京总部）   
mail：Eric@dwin.com.cn
服务热线：400 018 9008
总部地址:北京理工大学国防科技园5号楼18层
广州办事处:广州科学城科汇金谷三街4号C3幢10层
苏州办事处:苏州市相城区春申湖中路393号采莲大厦5楼501
武汉办事处:武汉市洪山区珞瑜路727号东谷银座1305   
生产基地:  湖南省桃源县漳江创新园创业大道8号迪文湖南科技园

bleughb, 2019-07-29:
I'd also add that both Creality and Wanhao have used the DGUS screens in their CR10SPRO and D9 printers. in addition, Wanhao have used the DGUSmini in the i3.

Download their code and see how they've implemented things - Creality seems to be the later optoin, based around Marlin 1.1.7, wheras wanhao hacked the guts out of 1.1.4 

Finally - may be worth contacting Sebastian on the ADVI3++ to discuss business. He's already working on some code, maybe he'll do business with you guys somehow?

InsanityAutomation, 2019-07-29:
I'm almost done porting all of the creality code into EXT UI. making a lot of improvements and basically rewriting large chunks along the way.

sjasonsmith, 2020-09-11:
@damo2641 is this feature request still valid? It seems like most modern displays are now using STM32 processors. It is hard for me to consider it worthwhile to invest time adding 8051 support.

sjasonsmith, 2020-11-10:
I'm going to go out on a limb and close this. I just don't see a world where it makes sense to add additional 8-bit HALs to Marlin. If somebody _really_ wanted it, I imagine they would need to add it themselves.

closed 2020-11-10 by sjasonsmith