MarlinFirmware/Marlin #14515: [BUG] Latest BF2 won't compile

oneofnine, 2019-07-05:
BTT SKR 1.3 board TMC2208's build attempt ..

```cpp
Compiling .pioenvs\LPC1768\src\src\module\planner.cpp.o
Compiling .pioenvs\LPC1768\src\src\module\scara.cpp.o
Compiling .pioenvs\LPC1768\src\src\module\servo.cpp.o
Compiling .pioenvs\LPC1768\src\src\module\stepper.cpp.o
Marlin\src\module\planner.cpp:1820:105: error: macro "MAX" passed 4 arguments, but takes just 2
block->step_event_count = MAX(block->steps[A_AXIS], block->steps[B_AXIS], block->steps[C_AXIS], esteps);
^
Marlin\src\module\planner.cpp: In static member function 'static bool Planner::_populate_block(block_t*, bool, const int32_t (&)[4], float, uint8_t, const float&)':
Marlin\src\module\planner.cpp:1820:29: error: cannot resolve overloaded function 'MAX' based on conversion to type 'uint32_t' {aka 'long unsigned int'}
block->step_event_count = MAX(block->steps[A_AXIS], block->steps[B_AXIS], block->steps[C_AXIS], esteps);
^~~
*** [.pioenvs\LPC1768\src\src\module\planner.cpp.o] Error 1
 [ERROR] Took 9.52 seconds
```

brunnels, 2019-07-05:
Looks like caused by b6546ea

Remove `#pragma once` in minmax.h and it builds

oneofnine, 2019-07-05:
Thank you for that @brunnels, that got it compiling again.

freydor, 2019-07-05:
Had the same problem compiling
git bisect led me to same solution as @brunnels 

[biscet.log](https://github.com/MarlinFirmware/Marlin/files/3363825/biscet.log)

closed 2019-07-06 by oneofnine