MarlinFirmware/Marlin #14516: [FR] Compilation problem

ainboa, 2019-07-05:
Hello, everyone, 

I come here because I have a problem. I'm trying to install Marlin 2.0 on a mks sbase. And when I compile I end up with a mistake. I attach a screenshot to show my concern. If someone has an idea I'm a taker;) 

Have a good day, everyone.

[image]

Ludy87, 2019-07-05:
Same #14515

Gerggg, 2019-07-05:
Me too :
Compiling .pioenvs\LPC1768\src\src\module\probe.cpp.o
Marlin\src\module\planner.cpp:1820:105: error: macro "MAX" passed 4 arguments, but takes just 2
block->step_event_count = MAX(block->steps[A_AXIS], block->steps[B_AXIS], block->steps[C_AXIS], esteps);
^
Marlin\src\module\planner.cpp: In static member function 'static bool Planner::_populate_block(block_t*, bool, const int32_t (&)[4], const float (&)[4], 
float, uint8_t, const float&)':
Marlin\src\module\planner.cpp:1820:29: error: cannot resolve overloaded function 'MAX' based on conversion to type 'uint32_t' {aka 'long unsigned int'}
block->step_event_count = MAX(block->steps[A_AXIS], block->steps[B_AXIS], block->steps[C_AXIS], esteps);
^~~
*** [.pioenvs\LPC1768\src\src\module\planner.cpp.o] Error 1

sl1pkn07, 2019-07-05:
fixed

thinkyhead, 2019-07-06:
Obvious issues are often fixed before we even read the report(ssssss).

closed 2019-07-06 by thinkyhead