MarlinFirmware/Marlin #14521: Bltouch V3

blueeagle69, 2019-07-06:
Hi.
I wasn't quite sure where to post this, so apologies if I messed up.
I have a Bltouch on order which I believe is the V3.
In Marlin 2.0, am I supposed to have a #define BLTOUCH_V3?
Because if so, I'll be damned if I can find it 🤷‍♂️
So if that is normal, should I just enable #define BLTOUCH as normal, and set #define BLTOUCH_SET_5V_MODE?
I'm running an SKR 1.3 mainboard/
Thanks.

Roxy-3D, 2019-07-06:
There was a BL-Touch v3 setting...   But it has been removed.   It isn't needed if you set what you want correctly:
```
  //#define BLTOUCH_FORCE_SW_MODE

  /**
   * Settings for BLTouch Smart 3.0 and 3.1
   * Summary:
   *   - Voltage modes: 5V and OD (open drain - "logic voltage free") output modes
   *   - High-Speed mode
   *   - Disable LCD voltage options
   */

  /**
   * Danger: Don't activate 5V mode unless attached to a 5V-tolerant controller!
   * V3.0 or 3.1: Set default mode to 5V mode at Marlin startup.
   * If disabled, OD mode is the hard-coded default on 3.0
   * On startup, Marlin will compare its eeprom to this vale. If the selected mode
   * differs, a mode set eeprom write will be completed at initialization.
   * Use the option below to force an eeprom write to a V3.1 probe regardless.
   */
```

blueeagle69, 2019-07-06:
Many thanks.
I thought I was going crazy.
Hey wouldn't be the first time :)

closed 2019-07-11 by thinkyhead