MarlinFirmware/Marlin #14524: [BUG] MIN/MAX Compiler warning incl. fix

dahuafschmied, 2019-07-06:
### Description
Marlin 2 Version from 6.7.19

Compiler warning: 
`n file included from Marlin/src/HAL/HAL_LPC1768/u8g/../../../core/macros.h:24,
from Marlin/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.c:37:
Marlin/src/HAL/HAL_LPC1768/u8g/../../../core/minmax.h:75: warning: "MAX" redefined
#define MAX(...)        _MAX_N(NUM_ARGS(__VA_ARGS__), __VA_ARGS__)

In file included from /Users/xxxx/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/lpc17xx_pinsel.h:44,
from Marlin/src/HAL/HAL_LPC1768/u8g/HAL_LCD_pin_routines.c:36:
/Users/xxxx/.platformio/packages/framework-arduino-lpc176x/system/CMSIS/include/lpc_types.h:149: note: this is the location of the previous definition
#define MAX(a, b) (((a) > (b)) ? (a) : (b))`

and the same for MIN

### Steps to Reproduce

Compile wich PlatformIO for LPC1768 on Mac

#### Additional Information
Bugfix:

change line 62 in minmax.h from
` #define MIN(...)        _MIN_N(NUM_ARGS(__VA_ARGS__), __VA_ARGS__)`
to
` #define _MIN(...)        _MIN_N(NUM_ARGS(__VA_ARGS__), __VA_ARGS__)`

and line 75 from
`  #define MAX(...)        _MAX_N(NUM_ARGS(__VA_ARGS__), __VA_ARGS__)`
to
`  #define _MAX(...)        _MAX_N(NUM_ARGS(__VA_ARGS__), __VA_ARGS__)`

sl1pkn07, 2019-07-07:
please fill a PR with the fix

greetings

thinkyhead, 2019-07-09:
Fixed by #14525

closed 2019-07-09 by thinkyhead