MarlinFirmware/Marlin #14529: Cannot compile for Ender-2: Make sure you have 'Sanguino' selected

probonopd, 2019-07-06:
### Description

Cannot compile for Ender-2

### Steps to Reproduce

Try to build for Ender-2 on Travis CI using

https://github.com/probonopd/Marlin/blob/a0d6a24f1ff4288c6bb6f1884b055e81f3c5798d/.travis.yml

**Expected behavior:** [What you expect to happen]

Compiles without errors

```
In file included from sketch/pins_SANGUINOLOLU_12.h:41:0,
                 from sketch/pins_MELZI.h:32,
                 from sketch/pins_MELZI_CREALITY.h:35,
                 from sketch/pins.h:207,
                 from sketch/MarlinConfig.h:40,
                 from sketch/G26_Mesh_Validation_Tool.cpp:27:
pins_SANGUINOLOLU_11.h:55: error: #error "Oops!  Make sure you have 'Sanguino' selected from the 'Tools -> Boards' menu."
   #error "Oops!  Make sure you have 'Sanguino' selected from the 'Tools -> Boards' menu."
    ^
exit status 1
```

#### Additional Information

I am building on Travis CI. I have set the board to `arduino --board Sanguino:avr:sanguino:cpu=atmega1284p --save-prefs`.

https://github.com/probonopd/Marlin/blob/a0d6a24f1ff4288c6bb6f1884b055e81f3c5798d/.travis.yml#L22

probonopd, 2019-07-06:
It works with `arduino --verify --board Sanguino:avr:sanguino:cpu=atmega1284p --pref build.path=build Marlin/Marlin.ino`.

thinkyhead, 2019-07-11:
Thanks for that information. I never was able to get the compile working by my poor guesses.

closed 2019-07-11 by thinkyhead