MarlinFirmware/Marlin #14530: [FR] Homing to a location

cdedwards, 2019-07-06:
Recently a power failure highlighted a problem with homing. About 2/3 of the way through a 35 hour  print a power failure occured.  What I needed to do is home in a specific location of the bed that had nothing on it. Without recompiling the firmware, there is no way to specify a location to home the Z. What I propose is  that if the X and Y in the G28 command have a value, the printer will  home in that particular location and not the hard coded location. Normal sequence is 
G28 XY ; home X and Y
G28 Z ; Home Z at hardcoded location

new sequence
G28 XY ; home X and Y
G28 X50 Y 250 Z  ; home Z at location 50/250 and not at hardcoded location.

AnHardt, 2019-07-06:
Do
G28 XY
G1 Xx Yy
G28 Z

cdedwards, 2019-07-06:
This will work if SAFE_Z_HOMING is not enabled. G29 would be a non-issue. The fade height will have taken care of the bed issues and you should be level across the entire print. I'm not talking about an issue that is close to the start of the print where restarting is more than likely the proper way to resume.

AnHardt, 2019-07-06:
A " bed needing G29" means a bed either not flat or not level or both. So with homing with a probe it depends where you prob. Homing once in the middle and once somewhere will result in a different z-height.

Did we recently change something?
Is  SAFE_Z_HOMING now used when not G28 or G28 XYZ. At least the last time i used G28 Z it did not move in XY. It was purposely designed to stay in place. Have not tried it for a long time. Someone may have changed that - but that would be an error.

cdedwards, 2019-07-07:
#if ENABLED(Z_SAFE_HOMING)

  inline void home_z_safely() {

    // Disallow Z homing if X or Y are unknown
    if (!TEST(axis_known_position, X_AXIS) || !TEST(axis_known_position, Y_AXIS)) {
      LCD_MESSAGEPGM(MSG_ERR_Z_HOMING);
      SERIAL_ECHO_MSG(MSG_ERR_Z_HOMING);
      return;
    }

    if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Z_SAFE_HOMING >>>");

    sync_plan_position();

    /**
     * Move the Z probe (or just the nozzle) to the safe homing point
     */
    destination[X_AXIS] = Z_SAFE_HOMING_X_POINT;
    destination[Y_AXIS] = Z_SAFE_HOMING_Y_POINT;
    destination[Z_AXIS] = current_position[Z_AXIS]; // Z is already at the right height

this definitely looks like it moves to Z_SAFE_HOMING points and then homes Z.

AnHardt, 2019-07-07:
[This one](https://github.com/AnHardt/Marlin/blob/8b02e68cb6009d3eb8695df26186e823bed31702/Marlin/Marlin_main.cpp#L2969-L3053) still has the logic i talked about.

Will see when and why this vanished.

AnHardt, 2019-07-07:
Got lost in https://github.com/MarlinFirmware/Marlin/pull/4789/commits/c84b14c77a5c1ae277a28e1dc0bd84094efc42ec part of #4789 , seemingly without discussion. Likely by error - not by purpose.

AnHardt, 2019-07-07:
In bugfux-2.0.x G28.cpp try: https://github.com/AnHardt/Marlin/pull/90
```diff
@@ -339,11 +339,14 @@ void GcodeSuite::G28(const bool always_home_all) {
       if (doZ) {
         #if ENABLED(BLTOUCH)
           bltouch.init();
         #endif
         #if ENABLED(Z_SAFE_HOMING)
-          home_z_safely();
+          if (home_all)
+            home_z_safely();
+          else
+            homeaxis(Z_AXIS);
         #else
           homeaxis(Z_AXIS);
         #endif
 
         #if HOMING_Z_WITH_PROBE && defined(Z_AFTER_PROBING)
```

That feature got lost that long ago (12 Sep 2016) some new features may be based on the new behavior.
Intensive testing is needed.

thinkyhead, 2019-07-11:
There's probably a good reason why `Z_SAFE_HOMING` always uses `home_z_safely`…. Because you cannot home Z just "wherever it happens to be now" when the probe is the only Z endstop. It will miss the bed, break the probe, pull apart the Z couplers, and set your livestock free.

robbycandra, 2019-07-15:
Thats why i prefer Z MAX homing.

closed 2020-07-01 by cdedwards