MarlinFirmware/Marlin #14531: Documentation request: WinAVR and Arduino IDE

sorjon02, 2019-07-06:
I am trying to set up the Arduino IDE to compile Marlin 2.0. I understand that this is a prerelease, and still under development, But I am willing to accept this, to have a look at the new version and maybe be able to contribute something to the Project.
Working computer for development: PC; OS: Microsoft Windows 10

So far I have:

1. Downloaded and installed Arduino IDE 1.8.9 C:\Users\[UserName]\Documents\Arduino is the 
Schetch dir.

2. Downloaded and installed WinAVR  (WinAVR-20100110-install) in c:\Users\[UserName]\Documents\Arduino\libraries\WinAVR

3. Downloaded support for Arduino Due and configured the Project for this CPU

4. Downloaded Marlin using "git clone https://github.com/MarlinFirmware/Marlin.git" to c:\Users\[UserName]\Documents\Arduino\Marlin

5. Attempted to compile Marlin 2.0, but with the error message "exit status 1
avr/io.h: No such file or directory" This is why I installed WinAVR

Despite my efforts during today, I have not been able to make the WinAVR Library available in the Arduino IDE. And yes: I have spent my time serching the Internet.

Can somebodey please spend a Little time documenting how to do this? If I can get it right and compile Marln 2.0 I'm willing to write a script to make the Marlin download and configure for compilation process automatically. (I have already began a script in PowerShell, alltrough so far it only checks that the correct versions of some SW is available.)

sl1pkn07, 2019-07-06:
why you not try to use platformio?

reloxx13, 2019-07-06:
no need for winavr

its included in arduino ide

move ur arduino folder to root of C:

[image]

2.0 compiled fine

dont use windows appstore arduino ide,
use latest arduino IDE for marlin 2 (cause of subfolders)

[image]

[image]

(ender 3 board)
[image]

sorjon02, 2019-07-07:
Problem solved! Arduino IDE is built on an early version of Gnu AVR/WinAVR, and does not support the avr/io.h include file and library.  My guess is that upgrading the support requires a new major version, and is probably not worth the effort, especially if your main interest is Marlin.

The best solution for you is probably to drop all support for Arduino IDE, and only support Atom/PlatformIO. If you state this on the website, it will save anyone interested in compiling Marlin 2 a lot of headscratching and effort.

closed 2019-07-07 by sorjon02

reloxx13, 2019-07-07:
dude, i just compiled marlin 2.0, i even posted screenshots of it

sorjon02, 2019-07-08:
I did read it! But I also spent two or three hours looking for the library avr/io.h in the Arduino IDE directory and subdirectories. Despite my effort, I could not find it. I decided to give up, and go for PlatformIO instead, and Marlin 2 compiled imidiately. So I go for Atom/PlatformIO now. I don't want to spend more time trying to make Arduino IDE work for me.

thinkyhead, 2019-07-09:
Atom is no longer being supported by PlatformIO.
Going forward you should all use Visual Studio Code instead.
MarlinFirmware/MarlinDocumentation#176

sorjon02, 2019-07-09:
I have now changed from Atom/PlatformIO to Visual Studio/PlatformIO, in accordance with  "Switch to PlatformIO IDE for VSCode #176 ". Sorry for this, but there has been a lot of invormation to take in, and I feel that it is fragmentetd. When it comes to Marlin development, it feels like everytime I has been presented with two choises I have managed to take the worst one. It's bad luck, but that's Life sometimes.
Best regards,

thinkyhead, 2019-07-11:
>  it feels like everytime I has been presented with two choises I have managed to take the worst one

Murphy's Law. 🤷‍♂