MarlinFirmware/Marlin #14534: [Bug] Online image converter buggy

VanessaE, 2019-07-07:
Foreword:  I'm putting this here in the Marlin issue tracker because I don't see a more appropriate place.

### Description

The online image converter located at http://marlinfw.org/tools/u8glib/converter.html has a couple of minor bugs, when generating a status screen logo:

1. The format/syntax of code block it creates is not Marlin-2.0 compatible:
  * The converter sets the image width as `STATUS_SCREENWIDTH`, but Marlin 2.x expects `STATUS_LOGO_WIDTH`.
  * The name of the array (?  what we call a table in Lua) is incorrect.  The converter sets it to 
`status_screen0_bmp` but Marlin expects `status_logo_bmp`.
  * Minor nitpick:  the converter is appending a redundant comma to the end of the bitmap data.
  * Not sure if it matters:  `#pragma once` is used in other example status screen images supplied with Marlin, but the converter does not insert that directive into its output.

2. The converter is inserting at least one space between items in the last line of output, where it should be inserting a comma.  This confuses the compiler, causing it to error out:
`  B00000000,B00000000,B00000000,B00000001 B11000000,`
(look between the 4th and 5th bytes)

3. On the converter page, the text "If this tool doesn't work in your browser, use the server-side tool at digole.com." links (at the end there) to a page that declares itself as deprecated, which then recommends users follow another link to an "online emulator", which, to put it politely, looks like something straight out of 1990, and seems not to be particularly useful...  I recommend just deleting this line of text from the page.

### Steps to Reproduce

Browse to http://marlinfw.org/tools/u8glib/converter.html and plug in a suitable image file, examine the output.

**Expected behavior:**

I should get a block of code I can copy/paste directly into _Statusscreen.h

**Actual behavior:**

I got a block block of code that required an enormous :wink: 4-line change to make it work.

#### Additional Information
<details><summary>_Statusscreen.h as generated by the converter</summary>

```
/**
 * Made with Marlin Bitmap Converter
 * http://marlinfw.org/tools/u8glib/converter.html
 *
 * This bitmap from the file 'marlin-prusa-i3-status-screen-logo.png'
 */
#define STATUS_SCREENWIDTH 40
const unsigned char status_screen0_bmp[] PROGMEM = {
  B00000001,B11111111,B00000001,B00000000,B00000000,
  B00000010,B00000111,B00000011,B00000000,B00000000,
  B00000100,B01010011,B00000011,B00000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B11001110,B00000000,
  B00000100,B10001001,B00001000,B01111010,B00000000,
  B00000010,B00000010,B00001111,B10010100,B00000000,
  B00000001,B11111100,B00011111,B11111100,B00000000,
  B00000000,B00000000,B00111111,B11111110,B00000000,
  B00000000,B00000001,B11111111,B11111111,B00000000,
  B00000011,B11111111,B11110111,B11111111,B10000000,
  B00000000,B00001111,B10111011,B11111111,B10000000,
  B00000000,B00000001,B11111011,B11101101,B11000000,
  B00000000,B00000010,B11111011,B11110101,B11000000,
  B00000000,B00000101,B10110111,B11110101,B11000000,
  B00000110,B00001110,B00001111,B11111110,B11000000,
  B00000101,B00000000,B00000000,B00000110,B11000000,
  B00000101,B01010101,B00110011,B10000110,B11000000,
  B00000110,B01100101,B01100100,B10000011,B11100000,
  B00000100,B01000101,B00110101,B10000010,B11100000,
  B00000100,B01000011,B01100010,B10000001,B11110000,
  B00000000,B00000000,B00000000,B00000001,B11010000,
  B00000000,B00001110,B00000000,B00000011,B10000000,
  B00000000,B00100010,B00000000,B00000111,B10000000,
  B00000000,B00000100,B00000001,B11111111,B10000000,
  B00000000,B00100010,B00000111,B11111100,B10000000,
  B00000000,B00100010,B00001100,B00000000,B11000000,
  B00000000,B00101100,B00000000,B00000111,B11000000,
  B00000000,B00000000,B00000000,B00000001 B11000000,
};
```
</details>

<details><summary>The corrected code</summary>

```
/**
 * Made with Marlin Bitmap Converter
 * http://marlinfw.org/tools/u8glib/converter.html
 *
 * This bitmap from the file 'marlin-prusa-i3-status-screen-logo.png'
 */

#pragma once

#define STATUS_LOGO_WIDTH 39

const unsigned char status_logo_bmp[] PROGMEM = {
  B00000001,B11111111,B00000001,B00000000,B00000000,
  B00000010,B00000111,B00000011,B00000000,B00000000,
  B00000100,B01010011,B00000011,B00000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B11001110,B00000000,
  B00000100,B10001001,B00001000,B01111010,B00000000,
  B00000010,B00000010,B00001111,B10010100,B00000000,
  B00000001,B11111100,B00011111,B11111100,B00000000,
  B00000000,B00000000,B00111111,B11111110,B00000000,
  B00000000,B00000001,B11111111,B11111111,B00000000,
  B00000011,B11111111,B11110111,B11111111,B10000000,
  B00000000,B00001111,B10111011,B11111111,B10000000,
  B00000000,B00000001,B11111011,B11101101,B11000000,
  B00000000,B00000010,B11111011,B11110101,B11000000,
  B00000000,B00000101,B10110111,B11110101,B11000000,
  B00000110,B00001110,B00001111,B11111110,B11000000,
  B00000101,B00000000,B00000000,B00000110,B11000000,
  B00000101,B01010101,B00110011,B10000110,B11000000,
  B00000110,B01100101,B01100100,B10000011,B11100000,
  B00000100,B01000101,B00110101,B10000010,B11100000,
  B00000100,B01000011,B01100010,B10000001,B11110000,
  B00000000,B00000000,B00000000,B00000001,B11010000,
  B00000000,B00001110,B00000000,B00000011,B10000000,
  B00000000,B00100010,B00000000,B00000111,B10000000,
  B00000000,B00000100,B00000001,B11111111,B10000000,
  B00000000,B00100010,B00000111,B11111100,B10000000,
  B00000000,B00100010,B00001100,B00000000,B11000000,
  B00000000,B00101100,B00000000,B00000111,B11000000,
  B00000000,B00000000,B00000000,B00000001,B11000000
};
```
</details>

Incidentally, here is the image I used, scaled up for visibility in this page:
[image]

original size:
[image]

(If you're interested in including that image/its code in Marlin as an example for i3's, feel free to do so.  It's derived from the original Marlin logo, so whatever license that uses, or CC-By-SA if you like)

AnHardt, 2019-07-07:
This tool could be used to produce Bootscreens, Custom Bootscreens, extruder graphics, bed, chamber, fan graphics, ...
So why is it producing more than the pure { data_field } and maybe the with.

On the other hand, there are enough working, compiling examples.
If a programmer is not able to transfer the produced file to his/her needs he really shouldn't try to change anything in Marlin. (The next time he will ask us to change his diapers.)

VanessaE, 2019-07-07:
@AnHardt, this issue is not for complaining about the existence of a tool or features it ought not have, it's about the tool not behaving as it was clearly intended.

I was about to go into a whole rant here, but suffice it to say I think you need to re-read the project's [Code of Conduct](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/code_of_conduct.md).

The whole purpose of writing a tool is to make some job easier, most often for the person who coded it. 
 That includes, when practical, formatting the output of the tool so that the user has to do the least possible amount of work to make the output usable for its intended purpose, however easy it may be, and however many existing, working examples there may be of the kind of output the hypothetical tool ought to produce.

That appears to have been the case for the online converter...  at least for a while.  Now it's just plain broken (i.e. the output it produces has no hope of being usable without being fixed-up).

I'm frankly glad it exists.  As easy as it would have been to convert the image manually, I really would rather not go back to the days of C64 BASIC programs with mountains of hand-coded `DATA` statements carrying images fonts, i.e. a time before we had reliable tools to gracefully manage stuff like that.

Ludy87, 2019-07-07:
this is not right position for your issues, see:https://github.com/MarlinFirmware/MarlinDocumentation

VanessaE, 2019-07-07:
Noted.  I'll move this issue over there so the rest can be tracked.

closed 2019-07-07 by VanessaE

VanessaE, 2019-07-09:
Since it's already in here, and it's independent of the actual issues in the converter, here's a revised version of the status image (because my OCD insists :stuck_out_tongue: ), just in case someone wants to use it:

<details><summary>_Statusscreen.h</summary>

```
/**
 * Made with Marlin Bitmap Converter
 * http://marlinfw.org/tools/u8glib/converter.html
 *
 * This bitmap from the file 'marlin-prusa-i3-status-screen-logo.png'
 */

#pragma once

#define STATUS_LOGO_WIDTH 39

const unsigned char status_logo_bmp[] PROGMEM = {
  B00000001,B11111111,B00000001,B00000000,B00000000,
  B00000010,B00000111,B00000011,B00000000,B00000000,
  B00000100,B01010011,B00000011,B00000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B11000111,B00000000,
  B00000100,B10001001,B00001100,B00111010,B00000000,
  B00000010,B00000010,B00010001,B11010100,B00000000,
  B00000001,B11111100,B00101111,B11111100,B00000000,
  B00000000,B00000000,B01111111,B11111110,B00000000,
  B00000000,B00000001,B11111111,B11111111,B00000000,
  B00000111,B11111111,B11111011,B11111111,B10000000,
  B00000000,B00001111,B01111011,B11011101,B10000000,
  B00000000,B00000001,B11111011,B11101101,B10000000,
  B00000000,B00000010,B01110111,B11101110,B11000000,
  B00000000,B00000101,B11101011,B11110110,B11000000,
  B00011000,B00001110,B00001100,B11100110,B11000000,
  B00010100,B00000000,B00000000,B00011111,B11000000,
  B00010101,B01010100,B11001110,B00000111,B11000000,
  B00011001,B10010101,B10010010,B00000011,B11100000,
  B00010001,B00010100,B11010110,B00000010,B01100000,
  B00010001,B00001101,B10001010,B00000011,B01110000,
  B00000000,B00000000,B00000000,B00000001,B11010000,
  B00000000,B00000111,B00000000,B00000011,B11000000,
  B00000000,B00010001,B00000000,B00001111,B10000000,
  B00000000,B00000010,B00000001,B11111111,B10000000,
  B00000000,B00010001,B00000111,B11111100,B10000000,
  B00000000,B00010001,B00001100,B00000010,B01000000,
  B00000000,B00010110,B00000000,B00000111,B01000000,
  B00000000,B00000000,B00000000,B00000001,B11000000
};

```
</details>

Image (scaled):
[image]

Normal size:
[image]

VanessaE, 2019-08-02:
@thinkyhead, is the above image of any use to you guys?  Never heard back about that.