MarlinFirmware/Marlin #14536: [BUG] BLTouch v3 is'nt triggering anymore on homeing.

Keichide, 2019-07-07:
So i just pulled the latest Marlin 2.x Version an hour ago and configured it for my Ender 3. Everything seems to work, the probe does a deploy and tries to home Z but never stops lowering the axis, so it does'nt trigger. I did'nt had any problem with 1.1.9 bugfix and also with an early 2.x version so it can't be the wiring. 

So.. Did i forget something in the configuration since you changed the whole bltouch implantion? I mean i know that i don't need the #define SERVO0_PIN 27 anymore since it's now part of pin_MELZI_CREALITY.h but any other option that worked before with the probe is also not working anymore.. 

I also already tried putting it in 5V Mode with the same result.

Keichide, 2019-07-07:
Okay.. It was sw_mode and 5v mode.. But you guys should really work a bit more on the description in the configuration_adv.h it's still to confusing.

closed 2019-07-07 by Keichide

InsanityAutomation, 2019-07-07:
Theres a reason 5v mode is set in the example configs... Feel free to write descriptions you find clearer and see if they fit. The probes themselves have become quite complex and confusing. after the hundreds of man hours across quite a few devs making everything work in the first place if the only complaint is descriptions, it went pretty well...