MarlinFirmware/Marlin #14542: [BUG] Custom _Bootscreen not working

benebrady, 2019-07-08:
### Description

### Steps to Reproduce

1. [First Step] Create a custom boot screen using the online bitmap converter
2. [Second Step] Compile firmware and flash it to the printer

**Expected behavior:** [What you expect to happen]
I expect the custom _Bootscreen,h to be displayed for the desired number of seconds. 
**Actual behavior:** [What actually happens]
The custom bootscrreen is not displayed on the LCD screen. It just shows some few lines of pixels on the screen and then it displays the Marlin bootscreen as expected.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
Here is a link to my dropbox with the configuration files and the _Bootscreen.h file that has failed. 
https://www.dropbox.com/s/798ild3l10f9jki/Ben%20E.%20Brady%20-%20_bootscreen%20bug%20report.zip?dl=0
[Ben E. Brady - _bootscreen bug report.zip](https://github.com/MarlinFirmware/Marlin/files/3367917/Ben.E.Brady.-._bootscreen.bug.report.zip)

Ludy87, 2019-07-08:
Select binary and try again

benebrady, 2019-07-08:
Tried it... Didn't work... Are you working on the problem or are you just shooting from the hip? I'm not the only one having problems with this.

benebrady, 2019-07-09:
've been really digging into this issue and I've found a few things... 
First, the bitmap conversion tool on the web doesn't generate the _Bootscreen.h correctly. If you generate a _Bootscreen.h file it does NOT work on 8 bit OR 32 bit controllers (I've tested MKS Gen L, MKS Gen 1.4 and SKR 1.3 boards using VSCode and PlatformIO) However, at the END of each line generated by the bitmap converter there appears to be an extra byte of binary code. If I remove that from each line and recompile the firmware, it DOES work fine on 8 bit controllers like the MKS Gen L and MKS Gen 1.4. It DOES NOT work at all on the SKR 1.3 (LPC1768 processor)
The attached .ZIP file contains TWO bootscreen.h files. Both of these files were generated by the bitmap conversion tool the only difference is I edited out the extra byte on the end of each line of the working _Bootscreen.h file (before the comment).  

[Example_Bootscreens.zip](https://github.com/MarlinFirmware/Marlin/files/3370732/Example_Bootscreens.zip)

Ludy87, 2019-07-09:
I apologize for asking a question to get an exact error picture. Nevertheless, I will try to help them, even if they drain stupid sayings.

benebrady, 2019-07-09:
Well, in the first place.. you didn't ask me any question... You simply told me to select binary and try again in the Bitmap converter. That's not a question... but at the same time I can understand that English may not be your primary language... I have provided more than ample information with regard to the problem and there appears to be more than one issue related to the problem. The first issue is that the web based bitmap conversion tool is NOT generating the proper code in order for it to work with 8 bit microcotrollers. If I edit the generated code I CAN get the 8 bit controller to display the custom bootscreen.. I have not been able to get the 32 bit microcontroller to display ANY custom bootscreen at all, but it DOES display the Marlin bootscreen. and "shooting from the hip" is not a 'stupid saying'... it's another way of saying "Are you just making a wild guess?", and not relying upon the facts and information people are providing to you. There IS definitely a problem related to the 32 bit processors and the implementation of the LCD libraries... and whatever has recently changed has also broken the functionality of the 8 bit processors.

gloomyandy, 2019-07-09:
For the image conversion tool, you may wish to follow (and comment) on the following...
https://github.com/MarlinFirmware/MarlinDocumentation/issues/233

Ludy87, 2019-07-09:
How bad my English is is totally uninteresting for the collective.
Do you want to kill me with that, or just leave a cool slogan?

mikekscholz, 2019-08-11:
This may or may not be relevant, but I was having the same problem using an image that was 128x64, but I trimmed the whitespace at the edges and tried again with a 120x64 image and it worked beautifully.

boelle, 2019-09-24:
@benebrady is the issue still there?

benebrady, 2019-09-24:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/MarlinFirmware/Marlin/issues/14542?email_source=notifications\u0026email_token=AKQG4Q2QGDRISDO7CBP4MDDQLINV3A5CNFSM4H62HS6KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7OP2QI#issuecomment-534576449",
> "url": "https://github.com/MarlinFirmware/Marlin/issues/14542?email_source=notifications\u0026email_token=AKQG4Q2QGDRISDO7CBP4MDDQLINV3A5CNFSM4H62HS6KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD7OP2QI#issuecomment-534576449",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Yes as of a week ago--Ben E. BradyNiederwald, Texas

boelle, 2019-10-21:
@benebrady  is the issue still there? please test with latest 2.0.x

boelle, 2019-10-30:
given that there has been no activity since 24 sep i will close this one as "fixed"

if multiple users can confirmed and we can get their configs we have a better chance

closed 2019-10-30 by boelle