MarlinFirmware/Marlin #14545: Emergency PAUSE and CONTINUE digital pin button

TorsionalWave, 2019-07-08:
Hi there, Does anyone knows if it is possible to stop the execution of the g-code ISTANTLY and restart if from exactly where it was?
It should be performed by external switch (Digital pin).
For example if a digital pin is put to GND it stops, when it is pulled UP again it restart the code.

The "ISTANTLY" condiction in needed because I would like to manage the FREQUENT and difficult to eliminate, plasma shut off phenomena in Plasma cutters.

I'm experienced with 3D printers that use Marlin and as I can remember that when you hit pause, you have to wait for the g-code buffer to empty, before to completely stop.

Maybe they are talking about this here?:
http://marlinfw.org/docs/gcode/M999.html 
or here:?
https://github.com/MarlinFirmware/Marlin/issues/11354
I'm not sure and most of all, can it be implemented into a BUTTON?

That is A VERY DIFFICULT FEATURE to develop, and needs a very deep knowledge of the marlin firmware but most of alla of the GRBL core from which it derives.

The reason is because it goes to hit the BARE METAL of the micro-controller.
To be able to perform the print reliably the GRBL team ( I Suppose) has organized the instruction into the SD card into G-code commands.
I think everyone or some of them are read from the SD card each time, translated into real step motor commands and executed.
I can't stop or restart a sigle G-code instruction in the middle of it blocking the read from the SD card, because the instruction (I suppose) are already been sent to the motor microcontroller and then they will go to motors.
I need to block the fast flow of information between micro-controller and motors. SD card maybe is already out of the game. 

Any Ideas? Thank you all guys.

oneofnine, 2019-07-08:
I too would like this feature, particularly when I'm trying new printer settings that don't quite work.

oneofnine, 2019-07-08:
I too would like this feature, particularly when I'm trying new printer settings that don't quite work as intended. I go to click the pause but it continues on for a second or two :(

phizz166, 2019-07-08:
I believe if you enable EMERGENCY_PARSER in configuration_adv.h, Marlin will respond to an M0 command with an immediate pause that ignores the buffer. Send M108 to continue.

I am not sure if Marlin natively has a way of generating custom commands on button activity, but something like OctoPrint can do it.

robbycandra, 2019-07-15:
You can get faster pause response if you make the planner buffer small.

DeeEmm, 2020-01-01:
I agree that Hold / Resume functionality should be able to be controlled via pins (external button / buttons).  

Whilst this may not really seem necessary for a 3D printer, there are many CNC applications using Marlin and on such installations using the display or a computer interface to access commands is not convenient. 

Hold / Resume buttons are standard controls on a CNC system.

Hold / Resume functionality should be instant (or as instant as-is safely possible).

---

I do however disagree with the terminology 'emergency' in this case. An 'Emergency Stop', is a very specific type of control on industrial machinery and so the terminology should not be used unless employed in the same manner. It is worth considering that those of us using Marlin on CNC machines are now crossing the boundaries between domestic 3D printers and small industrial CNCs. It would therefore be prudent to ensure that there is no confusion or ambiguity in terminology or functionality.

charlespick, 2021-05-19:
I believe this is implemented with `FREEZE_FEATURE` now

From Configuration.h:
 * Instant freeze / unfreeze functionality
 * Specified pin has pullup and connecting to ground will instantly pause motion.
 * Potentially useful for emergency stop that allows being resumed.

closed 2022-03-25 by ellensp