MarlinFirmware/Marlin #14547: [BUG] Both Z steppers move after Z_DUAL_STEPPER_DRIVERS is disabled

mersadk, 2019-07-08:
### Description

I have Core XY printer with two Z steppers.
Each stepper has it's own stepper driver (A4988). One is connected to Z port, other to E1 port.
I use MKS GEN v1.1 board. In `Configuration.cs` I've set `MOTHERBOARD` to `BOARD_RAMPS_14_EFB`.
In `Configuration_adv.cs` I've commented out `Z_DUAL_STEPPER_DRIVERS`.
I have a latest bugfix-2.0.x from git repository.

**Expected behavior:** When I try to move Z axes, I expect only one stepper to move, not both.

**Actual behavior:** When I issue command to move Z axes, both steppers will move.

#### Additional Information
I've figured out that there is an issue when trying to use G34 command (Z_STEPPER_AUTO_ALIGN). When I issue G34 command either both steppers on none will move (depending which stepper needs to compensate). It appers that stepper connected to Z port is "master" stepper, and the other is always cloned.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3370204/Configuration.zip)

RudyBenner, 2019-07-09:
Is there a reason did not use #define BOARD_MKS_GEN_13   
Line 441 in configuration_adv.h you did not define Z dual stepper drivers.
Line 601 you did not define Z_stepper_auto_align

[14547.zip](https://github.com/MarlinFirmware/Marlin/files/3372159/14547.zip)

mersadk, 2019-07-09:
@RudyBenner I didn't use BOARD_MKS_GEN_13 because I wasn't sure what is the difference between v1.1 and v1.3. With Marlin v1 I used BOARD_RAMPS_14_EFB and it worked fine (but I never tested this dual Z stepper issue).

Other options you mentioned are disabled for testing and that is a problem. Even when those are disabled, both Z steppers will move.

RudyBenner, 2019-07-09:
#define BOARD_MKS_GEN_13              1109  // MKS GEN v1.3 or 1.4
see the pin file for above, #include "pins_RAMPS.h"

RudyBenner, 2019-07-09:
Do M502 followed by M500 after flashing new firmware.
Sounds like old data in the eeprom messing things up.

mersadk, 2019-07-09:
@RudyBenner I'll try it later today when I get home and let you know the results. Tnx :)

mersadk, 2019-07-09:
@RudyBenner I think I found the issue. My board is MKS Gen-**2Z** v1.1. This is from one of a AliExpress product's description.

> MKS-Gen2Z is based on creating customer base MKS-Gen special customized products, especially for a single print head, **dual Z-axis 3D printer**

This board doesn't have E1 port, it's labeled as Z2. So probably Z1 and Z2 are mirrored in hardware.

I have some TMC2130 SPI drivers, I'll try with them. It should be possible to avoid this hardware restriction with custom pins.

RudyBenner, 2019-07-09:
Interesting.

mersadk, 2019-07-11:
I've switched to TMC2130 SPI but the problem remains.
After some investigation, it looks like TMC2130 in SPI mode still requires ENABLE/STEP/DIR pins. 
My understanding was that all commands are sent using SPI, but it looks like only configuration is done using SPI.

I'll try to reverse headers on stepper driver and remap them to some unused pings using dupont cables.
Hopefully this will work as a temporary fix until I get a better board.

mersadk, 2019-07-12:
At the end I was able to make it work by remapping E1 ENABLE/STEP/DIR pins to unused servo pins D4, D5 and D6.
To do it edit `pins_RAMPS.h`.
```
#define E1_STEP_PIN        5 // 36
#define E1_DIR_PIN         4 // 34
#define E1_ENABLE_PIN      6 // 30
```
Closing this issue.

closed 2019-07-12 by mersadk