MarlinFirmware/Marlin #14548: [BUG] Connecting Octoprint crashes Ender-3 board (bugfix-2.0.x)

sshockley, 2019-07-08:
### Description

Marlin 2.0.x-bugfix from Jul 5 crashes on Ender 3 Pro when connecting with Octoprint

### Steps to Reproduce

1. Boot printer normally
2. Press Connect in Octoprint interface
3. Ender/Marlin splash screen appears on LCD

**Expected behavior:** [What you expect to happen]
No reboots from connecting a USB device

**Actual behavior:** [What actually happens]

#### Additional Information
I upgraded from the stock firmware to try to install a BLtouch clone.  I couldn't get 1.1.x-bugfix to work with my sensor, so I tried 2.0.x-bugfix via git.  After spending some time trying to get 1.1.x to work, I do like what you're doing with 2.0.x.

At any rate, I'll be glad to get any logs or whatever you may need, I'll just need to know how to get them.  I did have to disable the sdcard so the bltouch would fit.  Diff of config is attached.  Thanks!

[Marlin20x.zip](https://github.com/MarlinFirmware/Marlin/files/3370587/Marlin20x.zip)

reloxx13, 2019-07-09:
this is how ATMega chips work/sync.

DerAndere1, 2019-07-10:
@sshockley : Please close this issue, if your printer can execute prints after the reset. As relox13 mentioned above, this is not a bug. The Atmega is expected to reset once during establishment of the connection.

sshockley, 2019-07-11:
Okay, I didn't realize that.  I don't believe the stock firmware did.

closed 2019-07-11 by sshockley

thinkyhead, 2019-07-11:
> I don't believe the stock firmware did.

It's actually a bootloader thing. Marlin has no control over stuff that happens before Marlin starts.

sshockley, 2019-07-11:
Thanks for the reply.  This is from the bootloader?  I think I'm missing something here, or I described the behavior badly.  I boot the printer normally; when I press Connect in Octoprint, the printer board reboots.  It also reboots if I press Disconnect in Octoprint.