MarlinFirmware/Marlin #14549: SKR 1.3 goes unto reset loop

TimboNZ76, 2019-07-09:
### Description

I upload basic marlin with board and serial ports defined and after it is uploaded and flashed to board it sends the board into a reset loop.

### Steps to Reproduce

1. Remove all parts from board and set it as it was when it came out of the box.
2. Download fresh marlin install from Github and make basic changes such as selecting the board and adding dummy thermistors just incase it throws a temp error.
3. Build Marlin 2.0 in Vcode with Platformio and copy bin file to SD card
4. Repower board and once it loaded press reset button 

**Expected behavior:** [What you expect to happen]
Expect board to go and stay connected to laptop and not reset

**Actual behavior:** [What actually happens]
Board goes into reset loop.  Marlin was showing as loaded on LCD when I did have it connected but would reset every 20 or so seconds

#### Additional Information

Will link configuration.h and configurationadv.h in comments as I am.on mobile.phone currently

TimboNZ76, 2019-07-09:
I'm expecting this to be a human interface error meaning I stuffed up somehow.

TimboNZ76, 2019-07-09:
I'm beginning to suspect there may be a bug in the temperature area as I set dummy thermistors 998 and 999 and when marlin starts up it has 0.0 for both temperatures where if should have 25 and 100. I'm wondering if watchdog is pulling me up on temperature issues

thinkyhead, 2019-07-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-07-09 by thinkyhead

manuelmarin, 2020-02-03:
yo tengo el mismo problema

thinkyhead, 2020-02-04:
No tienes el mismo problema. Un "boot loop" es un síntoma, no un problema.