MarlinFirmware/Marlin #14551: Dummy thermistors have no effect

TimboNZ76, 2019-07-09:
### Description

After setting dummy thermistors temperatures on LCD still show up as 0.0

### Steps to Reproduce

1. Edited config.h dummy thermistors
2. Reflashed board with new marlin
3. Reset board

**Expected behavior:** [What you expect to happen]
Expected thermistors to show temperatures from dummy thermistor

**Actual behavior:** [What actually happens]
Dummy thermistors don't make any difference thermistor temperatures still show o.o degrees celcius.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
https://www.facebook.com/groups/505736576548648/permalink/719663651822605/

gloomyandy, 2019-07-09:
Please post your configuration files so that folks can see exactly what changes you have made. It would also help if you specify what version of Marlin you are using for these tests (and if it is a bugfix version how recently did you pull the code).

TimboNZ76, 2019-07-09:
Sorry thought I had

TimboNZ76, 2019-07-09:
Downloaded marlin 2.0 literally today I will find my config files. I am using the SKR 1.3. I have also posed this question in Facebook groups and reprap forum.

TimboNZ76, 2019-07-09:
added config files

gloomyandy, 2019-07-09:
Your configuration files seem a little out of date (for instance the Configuration_Adv.h file contains old style SD card reader configurations). Note I'm not saying this has anything to do with your original problem or with the issue here, but you may want to consider starting with the default Marlin files and only making the minimal changes to them like board type, display etc. just in case! I am currently travelling so can't help much further on this I'm afraid.

gloomyandy, 2019-07-09:
Oh and just in case anyone else is investigating this it is worth pointing out that the user is seeing a boot loop. At the moment it is not clear if the boot loop is related to issues with thermal readings or if the issue reported here (thermal readings of zero) is some sort of consequence of whatever is causing the boot loop.

thisiskeithb, 2019-07-09:
Is this a duplicate of https://github.com/MarlinFirmware/Marlin/issues/14549?

TimboNZ76, 2019-07-09:
No it's a different bug but still by me. I gave up.on the resetting issue but have no idea why dummy thermistors don't register a temperature and that I thought was an actual bug. I apologised if again I got it wrong and should have asked everywhere else first.

thisiskeithb, 2019-07-09:
Hardware is the likely culprit. I just updated my SKR 1.3 to the latest commit (https://github.com/MarlinFirmware/Marlin/commit/ab99a807938b7fd8c5eafe2018b23ff876e765e6) and both the physical (5 & 11) & dummy thermistors work as intended.

TimboNZ76, 2019-07-09:
Fair call but how does hardware stop software based test thermistors. Checked entire board and nothing burnt out/shorted it popped.  Guess I will work with sler for a replacement. Thanks

TimboNZ76, 2019-07-09:
I will investigate further.

thisiskeithb, 2019-07-09:
[Here's my fork](https://github.com/thisiskeithb/Marlin/tree/Ender-3-SKR-1.3-TMC5160-Linear-Advance) (the readme has all the hardware details) for my SKR 1.3 setup if you want to compare.

TimboNZ76, 2019-07-09:
I will look it over and I may see where I went wrong.

gloomyandy, 2019-07-09:
So just to be clear are you saying that with the configuration files you have posted above your board is no longer resetting? The video in the link you posted to facebook seemed to be showing the board resetting.

TimboNZ76, 2019-07-09:
Well yeah the board does reset but I'm more curious about why I have no temperature readings using dummy thermistors now as no one at all has been about to solve the resetting problem.  If I put in the smoothieware bin and flash it the board stays going with no reset. But the reset not the issue first this post it was also about 0.0 temp readings lol.  Either way looks like I'm not gonna win this board battle yet lmao.

gloomyandy, 2019-07-09:
You can't just ignore the reset problem. The reset is almost certainly caused by a watchdog timeout. This means that at some point some operation is taking much longer than Marlin expects it to. 

The way that temperature readings work on Marlin is that what is displayed is a smoothed out version of what is actually read from the hardware. To do this Marlin needs to read a relatively large number of samples from the hardware before you will see any updates to the display. So if something is taking a lot longer than expected (which it probably is) then there is a good chance that Marlin may not have got around to updating the temperature displayed on the screen by the time the board resets. So it is very possible that whatever is causing the reset could also be causing the zero temperature readings you are seeing. 

The reset means that there is something wrong with one (or more) of...
1. Your hardware.
2. Your configuration.
3. Marlin.
At the moment there is no easy way to determine what exactly is causing the problem. A large number of people are using Marlin on the same hardware as you, so there is a good chance that there would be many more reports of the board resetting if it was a simple Marlin problem (it could still be a Marlin problem but possibly one that only shows up with a particular configuration). That is why I suggested above that you try a very minimal Marlin configuration. Take the sample Marlin config files and try only changing the board type and enabling your display. Leave everything else as is.

TimboNZ76, 2019-07-10:
I'm guessing it a hardware fault in my thermistor plugs as I turned off watchdog just too see and the whole marlin loads just no temperatures just 0.0 readings which I assume is why watchdog keeps resetting the board. Strangely adding the dummy thermistors does nothing temps still read 0.0  this is a fresh copy of marlin now pulled today and only has boars and LCD enabled with dummy thermistors set to 998 and 999 with 25 and 100 degree temperatures set.

TimboNZ76, 2019-07-10:
[Marlin 2.0 config files.zip](https://github.com/MarlinFirmware/Marlin/files/3375937/Marlin.2.0.config.files.zip)

New marlin files with dummy thermistors set.  Watchdog deactivated for now.

TimboNZ76, 2019-07-10:
Dummy sensors aside is there a way I could repurpose say the xyz max pins to run a thermistor temporarily as I heard someone reassigned the thermistors to new pins and suddenly they worked?

gloomyandy, 2019-07-10:
Try switching one of the thermistors back to the original setting and see what reading you get.
Do you have an actual thermistor that you can try?

TimboNZ76, 2019-07-10:
I tried an actual thermistor in both bed and hotend plugs with no change. Defined as 1 and 13 neither made any change.

gloomyandy, 2019-07-10:
When you say no change, do you mean that you just had a reading of zero all of the time? I thought that earlier on you had reported that with no thermistor you had a negative temperature reading?

TimboNZ76, 2019-07-10:
No regardless of what I do dummy thermistors or otherwise I get 0 degree temperatures. That's with thermistors attached, without thermistors attached or using dummy thermistors.  Sorry if I gave the impression I was getting negative temps I don't.  Basically 0.0 degrees is the standard for me.

TimboNZ76, 2019-07-10:
Has to be my board as my marlin compiles on another person's board perfectly.

closed 2019-07-10 by TimboNZ76