MarlinFirmware/Marlin #14556: [FR] Power-loss-recovery anlaog value

TB1405, 2019-07-09:
the mk3 power panic modul use for non fault condition a reference voltage from 0.8V and on error condition 0V. 
At the moment marlin supports only high and low pegel to detect a power outage. 
Would it be possible to implent the function for analog voltage ? 
for example
if(anlalogReadX<40){
poweroutage=1;
}

for sure it would be also possible to use a transistor to increase the signal but then you need a extra circuit board.

AnHardt, 2019-07-09:
Very unusual.
What happens when the input pin is drawn high by the processors internal pullup resistor?
If i remember correctly the pin at the Prusa is causing a interrupt. That would be impossible when read analog.

boelle, 2019-07-09:
also the einsy board itself cuts power to the bed and a few other things when the psu signal that mains has gone bye bye, it has to be done at hardware level as the bed would otherwise suck what power is left before the firmware can do anything

if you  dont trust me read the schematics for the einsy :-D

https://github.com/ultimachine/Einsy-Rambo/blob/1.1a/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.1a.PDF

boelle, 2019-07-09:
[image]

boelle, 2019-07-09:
[image]

to me it looks like that to power on bed or hotend logic signals need to come from both nAC_FAULT and the respective lines to turn on mosfets

again firmware is not in the picture here

boelle, 2019-07-09:
but what the einsy do provide is option to monitor power

[image]

this is used as part of the selftests, if power comes in but temp do not rise but it read temp it can ask user to check the power wires...

boelle, 2019-07-09:
but one thing is sure and that is analog is not used to detect power out, its done pure digital with either a high or low

if we are wrong please provide solid evidence that we are wrong :-D

AnHardt, 2019-07-09:
The important part is R68, a 10K pullup resistor.
So the output of the sensor could eiter to be a relay or on [open collector transistor output](https://en.wikipedia.org/wiki/Open_collector).

TB1405, 2019-07-09:
Alright you have me convinced.
No meaningful idea.  
Then I have to sell the modules again.  would have been too easy :D 
Thanks for the real detailed answere i think we can close this.

boelle, 2019-07-09:
only you can do that :-D

but yes, if you dont have an einsy board a simple relay or transistor is enough as sensor, i would pick a 230V relay so i can place it on the mains input

closed 2019-07-09 by TB1405

AnHardt, 2019-07-09:
???????????????????????????????
The mk3 power panic module **has** a open collector output. That's the reason for the low measured output voltage when high.
All it needs to use it with Marlin is either an external pullup resistor or a activated internal pullup resistor on the input pin. Both does not need witchcraft.

timmmoore, 2019-07-10:
One comment, I have a fairly old delta that I have ported marlin 2 to. They had both a power loss pin and power supply connected to an analog pin. Looking at their old version of marlin, the analog pin was being used to check the input power level and stopping the printer if it went too low. The reason was they shipped a PSU that wasn't powerful enough for bed, hotend and steppers. If you used all 3 at the same time, the power rail dropped and the hotend crashed into the frame (the level didn't drop enough for the 5V to drop). They had implemented a hack where they switched between hotend and bed and never turned both on at the same time but the voltage check was a safety guard. 
I didn't want to implement the switching logic, so I changed the PSU for a better one but I did implement the voltage level safety guard but just called kill if the level dropped too much. It hasn't triggered with the new PSU but with the old PSU it stops the printer before any really bad hotend crashes occur.
I dont know if this feature is of general interest? Note it does mean wiring the main power supply to a analog pin via a resistor divider.

TB1405, 2019-07-10:
@anhardt 
Thats a nice Idee i will check this later but i think in marlin isnt the Option to activate interanal pullup for the powerloss pin

timmmoore, 2019-07-10:
Hack to try, pick an endstop you are not using and will not get hit during homing, e.g. x_max if not delta
enable it with pull up and set power_loss_pin to the same pin.
Sanitychecks dont check that and it will get a pullup from endstop init. I dont think it checks it as an endstop unless it homes in that direction.

TB1405, 2019-07-10:
@timmmoore 
Very Nice idea i will try this.

AnHardt, 2019-07-10:
@timmmoore 
Thanks for the additional information.
We tried here to talk here about the Prusa version of power loss detection. Here the primary side (110V, 230V) of the PS is checked and one can use the remaining energy in the PS capacitors to do something useful. When waiting until the secondary side voltage is falling below a threshold it's almost to late to do anything useful. Once tried to compare the processors VCC to the internal reference of the processor (no additional hardware). Got about 10 usable assembler instructions on the AVR to do something. But what can you do in 10 ticks?

timmmoore, 2019-07-10:
@AnHardt, Not surprised there is no time, the delta board (its a Overlord Pro), looks at the 24V rail for both analog and the digital power loss pin but has a chargeable battery on the CPU power rail. That gives it enough time to run the power loss code.
I took a look at the MK3a code and they generate an interrupt from the power loss pin to be as fast as possible but they still need plenty of time from the PSU caps. I did notice they also have an analog pin looking at the 24V rail as well but its only used for info display.
I would have said the power loss using a pin is very dependent on hardware, either the way the MK3 does it or the way the Overlord does it are really the only options I can see. What Marlin does by saving each layer is probably the only generic way to handle it.
I think the analog power rail is more interesting for machines that have under speced PSUs. It catches that before the machine does bad things (too much), rather than for power loss recovery. I find its a comfort feeling if I am not there and something goes wrong, the delta does some crazy stuff when the power goes too low for the steppers but there is still lots of headroom for the 5V rail. I haven't had anymore of those issues since I upgraded the PSU but I connect the old PSU in occasionally to test it since its a comfort/safety feeling for me.

AnHardt, 2019-07-10:
> What Marlin does by saving each layer is probably the only generic way to handle it.

Generic - yes, but really bad. Up to a complete layer is repeated when recovering. But also the alternative, reacting on a pin is bad. Here we store the last read line and forget everything in the planner buffer - so we get a gap.