MarlinFirmware/Marlin #14557: [BUG] RADDS + ReprapDiscount Full Graphics Smart Controller shows blank LCD

ShadowOfTheDamn, 2019-07-09:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ShadowOfTheDamn, 2019-07-10:
Updates on the problem

at least I find good clue to solve the problem for now.
the problem is related to U8GLIB_ST7920_128X64_RRD

I managed to find that by changing lines 48 to 51 in ultralcd_DOGM.h to this:
``    #define U8G_CLASS U8GLIB_ST7920_128X64_4X ``
``   #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS     // Original u8glib device. 2 stripes, SW SPI ``
``    //#define U8G_CLASS U8GLIB_ST7920_128X64_RRD ``
``    //#define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS ``
and by commenting the line 209 in :
``  //u8g_dev_t u8g_dev_st7920_128x64_rrd_sw_spi = { u8g_dev_st7920_128x64_HAL_4x_fn, &u8g_dev_st7920_128x64_HAL_4x_pb, U8G_COM_ST7920_HAL_SW_SPI }; ``

the display works but it somehow a little bit slow.
I would be glad if any one with prior experience with Ug8lib library and St7920 driver help me out cause I have checked the delays before this solution with no success:
``#define ST7920_DELAY_1 DELAY_NS(400)``
``#define ST7920_DELAY_2 DELAY_NS(125)``
``#define ST7920_DELAY_3 DELAY_NS(440)``

ShadowOfTheDamn, 2019-07-11:
Anyone?

AnHardt, 2019-07-11:
This is not a support forum. We try to handle bugs and code improvements here.

We are bored by this problem. RADDS + ReprapDiscount Full Graphics Smart Controller was handled here several times (i guess). Please search.

ST7920 is an old, slow, buggy display controller. 

u8g-lib is a nice universal graphics library, but optimized to be universal and for saving memory, not for speed.

With writing our own u8g-device driver for the ST7920 we got some more speed.
If you really want to use the builtin u8g-device - you have to live whit its slowness.

Only our own device can be tuned by the custom delays. They are optimized by first setting ridicules high times, them shortening them, one by one, until you see pixel errors, then going back a step.

If the delays are not totally off you should see pixel errors.

Seeing nothing, usually is a pin or connection problem.

The problem seen in the picture is neither RADDS nor ReprapDiscount Full Graphics Smart Controller specific. It's a consequence of the [Picture Loop](https://github.com/olikraus/u8glib/wiki/tpictureloop) and programmers ignoring the consequences (Mainly: Don't change the screen content while in the loop, or at least if not in the first frame before drawing). It comes and goes almost always when some 'expert' tries to 'improve' the display code. Try to ignore it.

Currently it's worked on software SPI (again). It may help to use bugfix-2.0.x from a few days (weeks) before or later.

ShadowOfTheDamn, 2019-07-11:
Thanks a lot AnHardt. Then I will change my question. What type of LCD you propose with this kind of hardware I have RADDS + DUE.

AnHardt, 2019-07-12:
For most users ReprapDiscount Full Graphics Smart Controller  is 'fast enough' when used with the Marlin driver. LIGHTWEIGHT_UI does use a char mode for the statusscreen. It does improve the speed a lot when not in the menues.
About all other graphical displays either transfer less data, do that with a higher speed or both.  
Because of much less transferred data the character based displays like ReprapDiscount Smart Controller or the RADDS-display burn much less time. They transfer only ~80 bytes per screen update, vs. ~1 KB with the normal graphical displays and ~2KB with a ST7920.
The importance of the time to produce the display content shrinkend with the 32-bit processor.

ShadowOfTheDamn, 2019-07-16:
Thanks for the reply. Closing the issue. hope you would find some time to correct what you say you are supporting/developing. or remove it from your claims.

closed 2019-07-16 by ShadowOfTheDamn