MarlinFirmware/Marlin #14558: Conflict TCM2130 and PSU_AUTO_CONTROL

stormy-gh, 2019-07-09:
Problem:
Marlin hangs.

Steps to reproduce:

In Configuration.h enable:
#define X_DRIVER_TYPE  TMC2130
#define PSU_CONTROL
#define AUTO_POWER_CONTROL
In there is no connection with TCM2130 Marlin hangs.

Problem is that in restore_stepper_drivers() method stepperX.push() is called without any check if communication with TMC2130 is working.

Expected behaviour:
"TMC communication error" message.

teemuatlut, 2019-07-11:
Please provide configuration settings just like the issue template says.
Behaves just fine for me with SKRv1.3 running just on USB 5v.

stormy-gh, 2019-07-12:
I have board "DUE3DOM LE". It is version of DUE3DOM with SPI connected to stepsticks.
I attach schematic of the board.

Configuration changes I have made:

In _pins.h_
```
#elif MB(DUE3DOM_LE)
  #include "pins_DUE3DOM_LE.h"
```

In _boards.h_
`#define BOARD_DUE3DOM_LE              3099  // DUE3DOM LE for Arduino DUE`

In _Configuration.h_
```
#define MOTHERBOARD BOARD_DUE3DOM_LE
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define X_DRIVER_TYPE  TMC2130
#define Y_DRIVER_TYPE  TMC2130
#define Z_DRIVER_TYPE  A4988
#define PSU_CONTROL
#define AUTO_POWER_CONTROL  // Enable automatic control of the PS_ON pin
```

In this configuration Marlin hangs at startup.
When you comment out line _"#define AUTO_POWER_CONTROL"_ it working fine.

You do not need DUE3DOM LE to reproduce this behaviour - just try this config on standalone Arduino Due. My Arguino is hanging so hard, that I need to press "Erase" button to reprogram it again.

[Due3DomLE_Schemat.pdf](https://github.com/MarlinFirmware/Marlin/files/3385833/Due3DomLE_Schemat.pdf)

[marlin-bug.zip](https://github.com/MarlinFirmware/Marlin/files/3385827/marlin-bug.zip)

boelle, 2019-09-24:
@stormy-gh is the issue still there?

stormy-gh, 2019-11-03:
Yes. The issue is still there.

Quick & Dirty fix: I've commented out "restore_stepper_drivers();" on power.cpp line 127 and now everything is working good.

boelle, 2019-12-02:
@stormy-gh if not mistaken there has been some fix's to power control,

problem is that logic voltage needs to come on a tad before 12/24V when you deal with TMC's

give it a go

stormy-gh, 2019-12-02:
Informujemy, ze w dniach od 25 listopada - 18 grudnia firma jest nieczynna.

Panstwa list zostal zachowany, postaramy sie odpowiedziec najszybciej jak to bedzie mozliwe.

--------------------------------------------------------------

Please be informed that our office will be closed from 25th November to 18th December.

-- 
S-TECH Business IT solutions
http://www.s-tech.pl

boelle, 2019-12-24:
@stormy-gh its now past 18th december

 there has been some fix's to power control,

problem is that logic voltage needs to come on a tad before 12/24V when you deal with TMC's

give it a go

boelle, 2020-01-10:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-10 by boelle