MarlinFirmware/Marlin #14560: GTM32 PRO SD EEPROM Emulation does not work

AluOne, 2019-07-09:
### Description

SD EEPROM emulation from #14261 does not work since PR #14458

can't read or write settings

tpruvot, 2019-08-02:
Merged PR #14809 could have fixed your issue... Else, ensure you have at least one gcode file on the sdcard, else there is a problem to open the folder handle... in the shared base sdcard class

AluOne, 2019-08-05:
If I test a new version of today I can't compile with my board. 
I get a few errors.

```
platformio.ini
`board         = genericSTM32F103VE`
```
```
fastio_AVR.h
'DIOPC11_RPORT' was not declared in this scope
'DIOPC11_WPORT' was not declared in this scope
'DIOPC11_DDR' was not declared in this scope
'DIOPC11_PIN' was not declared in this scope
```
```
fastio_Due.h
'DIOSDSS_WPORT' was not declared in this scope
'DIOSDSS_PIN' was not declared in this scope
'DIOSDSS_WPORT' was not declared in this scope

```
```
HAL_spi_Due.cpp
'DIOPD2_WPORT' was not declared in this scope
'DIOPC12_WPORT' was not declared in this scope
'DIOPD2_PIN' was not declared in this scope
'DIOPC12_PIN' was not declared in this scope
```

```
spi_pins.h
'SDSS' was not declared in this scope

```
```
HAL_spi.cpp
#error "Invalid pins selected for hardware SPI"
```

```
HAL.h
'SerialUART1' was not declared in this scope
'afio_cfg_debug_ports' was not declared in this scope
'AFIO_DEBUG_SW_ONLY' was not declared in this scope

```

```
watchdog_STM32_F4_F7.h
'IWDG_HandleTypeDef' does not name a type; did you mean 'I2C_HandleTypeDef'?
identifier "IWDG_HandleTypeDef" is undefined

```
```
pins_GTM32_PRO_VB.h
#error "Oops! Select an STM32F1 board in 'Tools > Board.'"
'PC11' was not declared in this scope
'PC12' was not declared in this scope
'PD2' was not declared in this scope
'PC8' was not declared in this scope

```

boelle, 2019-09-24:
@AluOne so the issue is still there?

boelle, 2019-10-12:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-10-12 by boelle